motioneye-project/motioneye #1285: MotionEye not showing video

saleemsangi, 2019-05-19:
Can I please get some help with my new camera? I've an old camera with USB and it works fine but now I've bought a new camera [MakerFocus Raspberry Pi3 Camera IR ](https://www.amazon.com/MakerFocus-Wide-angle-Automatically-Day-Vision-Night-Vision/dp/B07DPNDP6R/ref=sr_1_3?crid=3HXDAHTCWMJK2&keywords=makerfocus%2Bpi%2Bcamera&qid=1558283617&s=gateway&sprefix=maker%2Bfocus%2Bp%2Caps%2C115&sr=8-3&th=1) and it's not showing the video. 

I think it's detecting the camera. Here is the picture when I add the camera from the UI.
[image]

Result for  lsmod command 

`Module                  Size  Used by    Tainted: G  
ipv6                  430080  0 [permanent]
brcmfmac              307200  0 
spidev                 16384  0 
brcmutil               16384  1 brcmfmac
snd_soc_bcm2835_i2s    16384  0 
snd_soc_core          188416  1 snd_soc_bcm2835_i2s
snd_bcm2835            32768  0 
snd_compress           20480  1 snd_soc_core
cfg80211              573440  1 brcmfmac
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm                98304  4 snd_soc_bcm2835_i2s,snd_soc_core,snd_bcm2835,snd_pcm_dmaengine
rfkill                 28672  2 cfg80211
snd_timer              32768  1 snd_pcm
snd                    69632  5 snd_soc_core,snd_bcm2835,snd_compress,snd_pcm,snd_timer
i2c_bcm2835            16384  0 
spi_bcm2835            16384  0 
fixed                  16384  0 
uio_pdrv_genirq        16384  0 
uio                    20480  1 uio_pdrv_genirq
i2c_dev                16384  0 
i2c_bcm2708            16384  0 
bcm2835_v4l2           53248  0 
v4l2_common            16384  1 bcm2835_v4l2
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  1 videobuf2_vmalloc
videobuf2_v4l2         24576  1 bcm2835_v4l2
videobuf2_core         45056  2 bcm2835_v4l2,videobuf2_v4l2
videodev              184320  4 bcm2835_v4l2,v4l2_common,videobuf2_v4l2,videobuf2_core
media                  32768  1 videodev`

I think I've the 20190427 version of OS, and using the Pi 3B.

Thanks

ilovelinux, 2019-06-20:
Can you give us the output of `ls /dev/video*`

saleemsangi, 2019-06-26:
My camera was faulty. Replaced the camera and it worked

closed 2019-06-26 by saleemsangi