motioneye-project/motioneye #1286: One camera overrides the feed of another camera. Same feed twice. One missing.

dylanparker, 2019-05-21:
motionEye version: 0.40
Motion Version: 4.0
OS Version: Debian 9.9

Not totally sure this isn't some oversight on my end...

I've narrowed down the issue to two specific cameras. I currently have only these two cameras installed in MotionEye and the I'm seeing the behaviour.
Both cameras are Trendnet IP Cameras Model TV-IP310PI
Both cameras are being accessed with AUTH via a URL like:
rtsp://HOSTNAME.lan/Streaming/Channels/1

If I disable one (toggle Video Device switch) then the remaining one correctly shows its video feed. When they are both enabled, then one of their feeds is displayed for both camera views.

To rule out conflicts with other NVR apps I've been evaluating, I just did a completely fresh install of a minimal Debian 9 and have only installed motionEye following the instructions for Debian on the wiki. The problem still occurs.

This happens identically in Chrome, Safari and Firefox.

Find attached a screencap of the duplicate feeds.
Also attaching the two camera-x.conf files.
Also attaching the motion.conf file.

Really enjoying setting this up so far. Hopefully can get past this hurdle. Please advise.

[image]

[camera-1.conf.txt](https://github.com/ccrisan/motioneye/files/3200401/camera-1.conf.txt)
[camera-5.conf.txt](https://github.com/ccrisan/motioneye/files/3200402/camera-5.conf.txt)
[motion.conf.txt](https://github.com/ccrisan/motioneye/files/3200403/motion.conf.txt)

dylanparker, 2019-05-21:
fwiw, I just modified the camera that had been added earlier (id 5) to match the settings of the newly added camera (id 1).

Now the diff of their conf shows just the naming and URL differences... and the duplicate feed persists. So, it doesn't appear to be settings related.

```
root@SENTRY:/etc/motioneye# diff camera-1.conf  camera-5.conf
12c12
< # @id 1
---
> # @id 5
44c44
< netcam_url rtsp://cam-backyard.lan/Streaming/channels/1
---
> netcam_url rtsp://cam-frontyard.lan/Streaming/Channels/1
63c63
< target_dir /var/lib/motioneye/Camera1
---
> target_dir /var/lib/motioneye/Frontyard
80c80
< camera_name Backyard
---
> camera_name Frontyard```

Continuing to investigate...

dylanparker, 2019-05-21:
I figured it out. It was my own mistake from a previous configuration misstep.
Earlier I had tried to duplicate camera settings for a new camera by copying the camera-N.conf to a new one and updating the ID. Some point soon after I started seeing that duplication and exported that config to do the Debian re-install. Then I re-imported that config.

So, @ccrisan, you've likely already spotted my error, but in copying the config I needed to update not only the ID but also the stream_port value. When the stream_port value is shared between camera entries, then `motion` goes into a shared-port mode which I'm guessing motionEye isn't prepared to deal with.

Based on my running around I wonder if motionEye can perhaps detect the repeated stream_port and either change it or somehow indicate to the user why the streams are broken.

Also, when adding many similar cameras, having some way of maintaining the wide range of similar settings would be great.

This can be closed, but I'll leave it open to make sure you see it.
Thanks again for the awesome frontend. It's exactly what I was looking for.

Kervinou, 2020-12-10:
Thanks
Very usefull for me 👍

velkrosmaak, 2021-02-17:
I had to remove some redundant thread-* files as well in order to fix the issue for me.