motioneye-project/motioneye #1288: Video toot fast

est-08, 2019-05-22:
Hello,
my problem is that whatever the video coding format used the playback of the videos is much too fast.

In the properties of the video file obtained the framerate is 1000 frames per second and the bit rate is 400 kbit / s.

Motioneye Version: 0.4
Motion Version: 4.0
VersionOS: Ubuntu 18.04

Help me please ...

jasaw, 2019-05-22:
Which video encoder are you using? Is it consistently recording like this? Have you tried a different video player?

est-08, 2019-05-23:
Hello and thank you for quick reply.
I tested with MPEG4, MSMPEG4v2, H.264 and Mastroka Video. the encoder is the default encoder of Ubuntu 18.04 I supose (ffmpeg it seems to me).
For the readers I used VLC and Médiaplayer under Windows and Totem under Ubuntu the problem is always the same.
The reading of the videos seems to be multiplied by 2.
In attachment a MediaInfo file of one of the videos.
[Vid.txt](https://github.com/ccrisan/motioneye/files/3212290/Vid.txt)

jasaw, 2019-05-23:
That sounds like PTS problem in the video file. I thought that has been fixed in motion software a long time ago. Try with a newer version of motion?

est-08, 2019-05-24:
Hello,
The version of Motion installed by default on Ubuntu 18.04 was 4.01.
So I made an update of Motion to download the .deb package on http://github.com/Motion_Project/motion/realeases
then execute "sudo apt-get install -f" and
"sudo dpkg -i bionic_motion_4.2.2-1i386.deb"

Motion went well in version 4.2.2
and the problem under MotionEye seems solved.
Thank you for your help.

closed 2019-05-27 by est-08