motioneye-project/motioneye #1289: cifs-utils in docker

Dinth, 2019-05-24:
Im running ccrisan/motioneye:master-amd64 image, but it seems that it doesnt include cifs-utils or smb support. Can you please add it to the container?

```
root@Motioneye:/# cat /proc/filesystems 
nodev	sysfs
nodev	rootfs
nodev	ramfs
nodev	bdev
nodev	proc
nodev	cpuset
nodev	cgroup
nodev	cgroup2
nodev	tmpfs
nodev	devtmpfs
nodev	configfs
nodev	debugfs
nodev	tracefs
nodev	securityfs
nodev	sockfs
nodev	dax
nodev	bpf
nodev	pipefs
nodev	hugetlbfs
nodev	devpts
	ext3
	ext2
	ext4
	squashfs
	vfat
nodev	ecryptfs
	fuseblk
nodev	fuse
nodev	fusectl
nodev	pstore
nodev	efivarfs
nodev	mqueue
	btrfs
nodev	autofs
nodev	rpc_pipefs
nodev	overlay
nodev	nfs
nodev	nfs4
nodev	aufs
```

Dinth, 2019-09-08:
sorry for refreshing the old issue, but is there anything what can be done to enable SMB support in the official motioneye container?

kleini, 2019-09-08:
I do not like to add cifs-utils to the docker container. This does not really make sense for me. I would mount some Samba volume on the underlying host operating system and mount it then into the container. What is your use case to need cifs-utils inside the container?

Dinth, 2019-09-08:
Hmm i can see your point here but for me seems like a security threat. The computer im running motioneye is my network firewall & IOT machine. There's quite a few of potential vectors of attacks here and i dont really want to expose my storage to whole machines. Docker provides at least some kind of isolation between different apps & services and allows me to configure each service to only see bits required by it.

kleini, 2019-09-08:
If somebody is able to break into your host machine and see the Samba filesystem there, he will be also able to break into the container running motioneye.
In my eyes Docker images & containers intention is to transport a single application - motioneye in this case - and all the necessary dependencies for exactly only this application. This is my understanding of Docker images and how to run containers. Optimized for a single application.
The wiring into all other services or especially other filesystems in this case, is a matter of the machine hosting the container.
Please see my point regarding your fear regarding your mentioned attack vector. If somebody is able to access the filesystem on your host, he will be able to access everything, that is inside the container, too.

JamBalaya56562, 2026-11-10:
Closing, as this is not planned: the image carries motionEye and what it needs, as kleini explained. A Samba share can be mounted on the host and bound into the container, for example at `/var/lib/motioneye` for the recordings.

closed 2026-11-10 by JamBalaya56562