motioneye-project/motioneye #1291: Manual IP address (static IP), the displayed date is 1970-01-01

aclemfr, 2019-05-25:
Hello,
I use the version "motionEyeOS 20190427" on raspberry pi zero.
By setting a manual IP address (static IP), the displayed date is 1970-01-01
at the location of 2019-05-25.
I have 3 cameras with raspberry pi zero. I have the same problem on these 3 cameras.
In automatic mode (DHCP) no problem displaying the current date.
Cordially.
-------------------------
Bonjour,
J'utilise la version "motionEyeOS 20190427" sur raspberry pi zero.
En mettant une adresse IP manuel (static IP), la date affiché est 1970-01-01 
au lieux de 2019-05-25.
j'ai 3 cameras avec raspberry pi zero. j'ai ce même problème sur ces 3 cameras.
En mode automatic (DHCP) aucun problème d'affichage de la date actuel.
Cordialement.

zagrim, 2019-06-12:
Hello. As you say, you are using motionEyeOS, so it might be better to ask this from [motionEyeOS site](https://github.com/ccrisan/motioneyeos).
Anyway, my guess is that you've likely got have an issue with routing and/or firewalls that prohibit traffic to the NTP server from the static IP range you are using, which would be a more generic network configuration issue and not related to this application at all.[](url)

dtokez, 2019-06-15:
Check your gateway IP, I had the same issue till I checked mine and set it correctly

JamBalaya56562, 2026-11-08:
Closing as answered above: with a static IP the Pi could not reach a time server, usually because the gateway was not set correctly, which fixed it for dtokez. The Pi Zero has no clock of its own, so it starts at 1970 until it gets the time over the network. This was on motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562