motioneye-project/motioneye #1292: change date format

redbaron101, 2019-05-26:
hi there,

Where does it get the date format from? It's currently YYYY/MM/DD. I want it to be DD/MM/YYYY.

motionEye (motionEyeOS 20190427 v4.2.2) is already installed on my raspberry pi3- tried looking for config.py but can only see the compiled ones (in /usr/lib/python2.7/site-packages/motioneye).

Thanks,

ccrisan, 2019-05-27:
Where exactly do you want this format to be changed?

redbaron101, 2019-05-31:
when i run motionEye from the browser, the camera shows with the date and time in the bottom right hand corner.  This is fine, but the date format is currently in YYYY/MM/DD.  Would prefer the format to show in DD/MM/YYYY .  Also I've set the time to GMT, but the time also shows 1 hour slow too?

redbaron101, 2019-06-02:
found it - text overlay

closed 2019-06-02 by redbaron101