motioneye-project/motioneye #1302: Motion Detection doesnt seem to be working

angelnero, 2019-06-03:
Set up a fresh install of Ubuntu 18.04.02 Bionic and installed MotionEye according to the installation wiki.

I have 8 IP cams set up and appear to be working.

Cameras are setup to use networked storage on a windows 10 based system.  Testing connection reports successful.

Motion detection settings are really aggressive right now in an attempt to pick up all movement.

frame change threshold = .05
auto noise = on
light change = off
motion gap = 10
before and after = 5
min motion frames = 20
mask = off

for some reason no motion is being detected.  on the off chance i see a red frame around one of the camera feeds (to indicate motion), no recording is posted to the networked file storage (even though test connections to the path are successful).

when i issue a sudo systemctl status motioneye on the motioneye machine it shows it is active but posts a few "warning: connect error on fd 24: econnrefused"

I am not sure what this warning means or if its even relevant.

Looking for some help here.

jasaw, 2019-06-04:
What frame rate are you running? Try reducing the min motion frames to less than 1 second. If you run 10 fps, set min motion frames to 10.
Try reducing the motion threshold even further.
Turn on "show debug frames"

brunetton, 2020-05-29:
Maybe this is related to #1774 ? Setting frame change threshold to very low values can disable motion detection

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up to jasaw's suggestions: Minimum Motion Frames at 20 needs 20 frames in a row with motion, which at a low frame rate is several seconds, and Show Debug Frames shows what Motion sees. Please open a new issue with `motion.log` if detection still does not trigger with a current release.

closed 2026-11-10 by JamBalaya56562