motioneye-project/motioneye #1303: Camera device/type changed following power cut

MalFrance, 2019-06-05:
Hi,

I had a Pi1b running motioneye with a NoIR pi camera attached quite happily (set and forget :) for some time until a power cut.
The Pi recovered without incident but it seems that while motioneye is nominally running, the camera setting has changed and so no video operation

A couple of things i can see:
The motion version is not displayed, but not sure if that wasn't always the case anyway.
[image]

If i simply try and run motion from the shell i get
[image]

The current camera setting is now:
[image]
but my notes show that it should be:
[image]
Trying to add a new V4L2 camera shows no cameras are detected.

bcm2835-v412 is still shown in /etc/modules, and if i run modprobe bcm2835-v412 from the shell nothing seems to happen.
Raspistill still works fine.
Rebooting doesn't do anything, but if i try to do apt-get update then i get a "segmentation fault" 

Seems like motion - at least - is broken, if not raspbian itself?
I could rebuild it all from the OS up again, but i've not really got the time before i go away at the weekend.
Any suggestions on what i can do to try recover this?

Cheers,

Mal

CRImier, 2019-06-05:
I would suggest an OS reinstall, this sure doesn't seem like a Motioneye problem if `apt-get` is not working (which is an indicator of something being wrong with the OS. More likely, it seems that some of the shared libraries on your system got corrupted after a power cut. This also most likely means you should also get a different SD card, one that's brand name (and purchased from a reputable shop where likelihood of getting a fake is low).

The reason I'm suggesting an OS reinstall is because it's quite easy to flash and set up Raspbian, compared to however much time you can potentially spend while trying to fix your current install so that it works - and if you can't make it work, you will be reinstalling either way.

Otherwise, I would suggest reinstalling the `libsqlite` thing (from the motion screenshot), i.e. by copying it from a different Raspbian install with the same libsqlite3 version - that is, if you have one available. `apt-get` doesn't link to `libsqlite`, however, which is what leads me to believe that the problem is large enough to suspect it might turn out to be a timesink larger than any reinstall could be. If you're determined to fix it, however, finding corrupted files and replacing them one-by-one will work, too.

MalFrance, 2019-06-05:
Thanks for the quick reply, i had a feeling that a rebuild might be the best/only solution.
I have a small supply of SD cards here so i'll use another - they're all 16Gb Sandisk Extremes which i use all over (Camera, Pis, GPS, car audio etc) and even though i don't remember who i got them from then i took a little time to choose a vendor and paid a little more to avoid fakes.
Once done and working i'll make sure to grab an image as well, to hopefully avoid similar problems again in the future.

Cheers,

Mal

CRImier, 2019-06-05:
Yep, grabbing an image sounds like a good measure in case this same thing happens in the future! Please don't forget to close this issue if your problem is solved.

closed 2019-08-12 by MalFrance