motioneye-project/motioneye #1304: Error on playback iOS

mix91, 2019-06-06:
Hi,

I have set my movie files to MP4. On Windows and Android I can playback all movies. But on iOS and Macbook I always get "Media format unsupported or otherwise unavilable/unsuitable for playing.". I have tested several browsers for iOS, but there is always the same problem.
May someone could help me?

Thank you!
Greetings Michael

bob-lee, 2019-06-14:
@mix91 I have iPhone5s and can play movies on Chrome or Safari browsers. On motionEye settings, Movie Format is set to `H.264/OMX (.mp4)`.

ilovelinux, 2019-06-20:
@mix91 what browser are you using? What OS is running MotionEye? What if you download the movie, does it play?

nicholastripp, 2019-08-17:
I have the same issue. I've tried H.264, HEVC, and QuickTime as formats for the video. On my Mac they'll all play with VLC, but not with QuickTime. Which means they don't play in Safari. And that means they don't play on iOS. 

motionEye Version | 0.39
Motion Version | 4.2.2
OS Version | Ubuntu 16.04

mwidick, 2020-05-25:
Add me to the list! Same problem. To make it stranger sometimes H.241 works in Safari and then suddenly stops being recognized on iPad and iPhone??? Seems to look like an MPEG. VLC will recognize but not Safari.

mwidick, 2020-05-25:
I am on RP0 to a RP4 server.

nicholastripp, 2020-05-25:
You should try installing a newer version of Motion. The version included in the Raspbian repositories is old. Installing the latest got most everything working for me.

hasse-basse, 2022-06-15:
Iphone has constraints when it comes to resolution of mp4, max is 1080p. For me it was as simple as setting the Movie Quality to 75% instead of 100%, et voilà, I could play the movies on my iPhone.

When it comes to Movie Format, the only mp4 that can be used is H.264 (.mp4) The other two H.264 options, with more letters in the name, can often not be used, it seem to be an issue about what OS you are using and what Codecs that comes with the OS. I use Raspberian Buster and H.264 (.mp4) is the only option that works for me. This also affects the Time laps movie making.

JamBalaya56562, 2026-11-07:
Closing as a duplicate of #1206: at 100% movie quality Motion writes H.264 that Safari and iOS cannot play, and lowering the quality to 75% fixed it for hasse-basse above.

closed 2026-11-07 by JamBalaya56562