motioneye-project/motioneye #1305: Network Camera not working

magooz, 2019-06-07:
Hello,

I have a lot of problems to commission my network camera.
I can add It but the stream doesn't work. However it works with vlc.
After a few hours trying to solve the problem, I decide to post this issue.

HIKvision: DS-2CD2645FWD-IZS    V5.5.80 build 180911

MotioneyeOS - 20190427
rtsp://192.168.1.246:554/streaming/channels/101
resolution and fps are the same on both systems.

Any ideas?

[motion (1).log](https://github.com/ccrisan/motioneye/files/3264866/motion.1.log)
[dmesg.log](https://github.com/ccrisan/motioneye/files/3264867/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneye/files/3264868/boot.log)
[messages.log](https://github.com/ccrisan/motioneye/files/3264869/messages.log)
[motioneye (1).log](https://github.com/ccrisan/motioneye/files/3264870/motioneye.1.log)

ccrisan, 2019-06-11:
Does it work with plain Motion? See [this](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) if you're wondering how to run Motion standalone on a motionEye setup.

JamBalaya56562, 2026-11-08:
Closing, as ccrisan's question whether the camera works with plain Motion went unanswered. This was on motionEyeOS 20190427; the RTSP stream is read by Motion, so if a current Motion still cannot show a stream that VLC plays, please report it to [Motion](https://github.com/Motion-Project/motion) with `motion.log`.

closed 2026-11-08 by JamBalaya56562