motioneye-project/motioneye #1306: Typo on Wiki

hamiltont, 2019-06-09:
Wiki says this (https://github.com/ccrisan/motioneye/wiki/Configuration-File) : 

```
motionEye uses a configuration file (usually called motioneye.conf) that allows customizing your setup (including paths, logging, various timeouts and other options that cannot be modified through the UI). The default location for this file is /etc/motioneye/motioneye.conf and can be changed using the -c command line switch.
```

On my new RPi Zero W installation this is untrue. I have /etc/motioneye.conf which points to `/data/etc/` which then has a number of `conf` files (only some of these are motioneye specific). 

Is this a little typo?

ccrisan, 2019-06-11:
You're on [motionEyeOS](https://github.com/ccrisan/motioneyeos/). This is motionEye.

closed 2019-06-11 by ccrisan

hamiltont, 2019-06-11:
D'oh 🤦