motioneye-project/motioneye #1308: Clean Cloud not working

Bryan241, 2019-06-11:
Hello
I’m using the Google Drive upload service for my motioneye and I enabled the « Clean Cloud » feature. But there’s some problem. At first sight it seems to work : files get removed daily but in reality they dont get deleted they just revoke my ability to watch them and so they still take place in my drive but I cant see them unless I check the largest files (yes my bin is empty)

Can you help me ?
Thanks !

ccrisan, 2019-06-11:
@Bryan241 can you please attach your logs here?
@bob-lee any ideas?

Bryan241, 2019-06-11:
> @Bryan241 can you please attach your logs here?
> @bob-lee any ideas?
How can I send you my log, but the problem come from the Google Drive clean cloud feature because on my raspberry pi, folders are deleted correctly and are no longer on my disk

bob-lee, 2019-06-11:
@Bryan241 your log doesn't show any activity around gdrive cleanup feature. Can you provide longer logs and `camera-1.conf` at `/etc/motioneye` so that I can get some clue? Below is taken from logs of my rpi for example:
>May 24 07:44:23 raspberrypi meyectl[214]:     INFO: cleanup started
May 24 07:44:23 raspberrypi meyectl[214]:     INFO: wsswitch started
May 24 07:44:23 raspberrypi meyectl[214]:     INFO: tasks started
May 24 07:44:23 raspberrypi meyectl[214]:     INFO: mjpg client garbage collector started
May 24 07:44:23 raspberrypi meyectl[214]:     INFO: server started
Jun 02 08:22:56 raspberrypi meyectl[214]:     INFO: gdrive: found 0/2 folder(s) in local/cloud
Jun 02 08:22:56 raspberrypi motion[518]: [1:ml1:Kitchen Doo] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has be
Jun 02 08:22:58 raspberrypi meyectl[214]:     INFO: gdrive: deleted a cloud folder '2019-05-24'
Jun 02 08:23:01 raspberrypi meyectl[214]:     INFO: gdrive: deleted a cloud folder '2019-05-23'
Jun 02 08:23:01 raspberrypi meyectl[214]:     INFO: gdrive: deleted 2 cloud folder(s)

ccrisan, 2019-06-11:
Setting debug logging (in motioneye.conf) could help.

ccrisan, 2019-06-11:
https://github.com/ccrisan/motioneyeos/issues/1924 may also be related.

Bryan241, 2019-06-11:
Here's my `camera-1.conf` and I'll wait tomorrow for the clean drive 
https://hastebin.com/ewocuseqir.pl

Bryan241, 2019-06-11:
I manually took a snapshot to see if its written in logs but there's nothing written in, but my snapshot is correctly uploaded on my drive

bob-lee, 2019-06-12:
@Bryan241 I normally run below to get a log file:
>sudo journalctl -u motioneye > my.log

Bryan241, 2019-06-13:
So here's the log that show this : https://hastebin.com/apamuxeseh.nginx

But, as I said, at first sight the folder "2019-06-11" disapeared : http://prntscr.com/o1f6oo
Bullshit ! If I check my trash, I can search the largest files on my drive AND ! 
I can find some video from that folder ! : http://prntscr.com/o1f7t2

So... I check that folder... and here it is, I dont know where but it exist : http://prntscr.com/o1f89t

So lets check the folder's activity  : at 03:37 it gets deleted and at 03:38 I got my access revoked : http://prntscr.com/o1f9ic

bob-lee, 2019-06-13:
@Bryan241 Google rest api doc says:
>Permanently deletes a file by ID. Skips the trash.  

So the folder deleted by motioneye shouldn’t appear in the bin in theory..
Can you please monitor for a few days and advise if that happens repeatedly?

Bryan241, 2019-06-13:
The folders doesnt show up in my trash or even anywere, but still take place on my drive ! I’m monitoring this since I use motioneye (1 month ago) and it happens multiple times that my drive was full of Space, and yet, nothing in trash... its like they disapear but still take place ... somewhere because I had to check the « largest files » on my drive to delete them and gain Space and files I deleted was previous videos and photos drop motioneye that were supposed to be deleted

Bryan241, 2019-06-13:
I dont know if you understand but if you want i Can make a Video or you Can use AnyDesk and i’ll show you everything

bob-lee, 2019-06-13:
I think this is more of how Google Drive works. To my knowledge, a deleted file stays in `Trash` for some days, then moved to `Post-Trash` and stays there for some days then really-permanently deleted. So there is a chance to get back the deleted file during that period of time as you observed.

Bryan241, 2019-06-13:
> I think this is more of how Google Drive works. To my knowledge, a deleted file stays in `Trash` for some days, then moved to `Post-Trash` and stays there for some days then really-permanently deleted. So there is a chance to get back the deleted file during that period of time as you observed.

by the doc of Google Drive, they said that a file that goes to trash stay there for ever if we dont purge it manually 

I think you should edit your script to not bypass the trash and manually (or by a App Script) purge the trash

by this way, the file get really deleted and dont take place on drive "virtually" because when I look at the location of folders its like they are their own location somewhere in the root folder like an hidden folder

Bryan241, 2019-06-13:
Here's an explication in video https://puu.sh/DFzaP/535167f73a.mp4

bob-lee, 2019-06-13:
@Bryan241 not sure which google document you are referring. Clean cloud feature is to save users effort to visit Google drive periodically and remove old files to prevent their usage quickly reaching 15g quota limit. If you are patient to wait for Google drive to permanently delete them, they will be really gone and wouldn’t take up your usage. Below is the api that motionEye is using to delete files on drive:

https://developers.google.com/drive/api/v2/reference/files/delete

Bryan241, 2019-06-13:
But I reach the 15go quota on a month and then my raspberry pi cannot upload any files :/ 
Files must be deleted faster, when I manually delete them (on web or via AppScript) they get deleted instantly and dont take space in my storage any longer

Bryan241, 2019-06-14:
@bob-lee look at the I just posted, you Will understand. My folders are supposed to stay one day and then they are deleted but the day before which was supposed to be deleted is just « Hidden » or deleted but still take place on disk as you can see in the vidéo because I Can refind the 2019-06-11 folder (I don’t know where) (the folder is supposed to be in Camera/2019-06-11 but when they got deleted it seems they move in root and stay « hidden » because they still take place and I must delete them like I do in the vidéo to have free space

bob-lee, 2019-06-14:
@Bryan241 I saw your video thanks. I inspected my drive, found that among the deleted-yet-taking-space files, oldest one was around 2 month ago. From that, I assume that below is happening to the file deleted by `clean_cloud` feature in current Google Drive:
- will not appear on drive nor trash but taking up space
- can be restored by `Remove` menu to move it to trash then `Restore` menu
- after 2 month, Google Drive will delete permanently, frees up space   

So available options would be:
- if your storage has much room, you can leave `clean_cloud` feature on and do nothing and you have a chance to restore the deleted file if it was deleted within last 2 month
- if you need to free up space and are sure that you wouldn't need it again, `Remove` to move it to trash then `Delete forever` menu
- or turn off `clean_cloud` feature and maintain media files manually, e.g. `Remove` and `Delete forever` at your own risk

Deleted files lingering over 2 month makes sense to me as it can happen by any chance that you need to restore old video after some incident.

Bryan241, 2019-06-14:
Is there a way to delete folder older than 1 day automatically with Google App Script ?

Bryan241, 2019-06-14:
Up : I just try the API from Google (https://developers.google.com/drive/api/v2/reference/files/delete) and my folder got deleted correctly and no longer took space in storage, so I think the problem comes from your script

Bryan241, 2019-06-14:
```    def _delete_child(self, folder_id, child_id):
        url = '%s/%s/children/%s' % (self.CREATE_FOLDER_URL, folder_id, child_id)
        response = self._request(url, None, None, True, 'DELETE')
        succeeded = response == ""
        return succeeded```

You're doing this to delete a file on cloud but why dont you do this

```    def _delete_child(self, folder_id, child_id):
            service.files().delete(fileId=child_id).execute()
             ...```

Bryan241, 2019-06-14:
[image]

bob-lee, 2019-06-15:
@Bryan241 you are welcome to contribute as you have much interest on this project.

Bryan241, 2019-06-16:
So I disabled Clean Cloud feature and I’ve done something with Google AppScript 
It runs this every day :
```
function chkDateOfEveryFolder(parentFolderID) {
  var d,dateCreated,diffInMS,folder,folders,
      msInTimePeriod,nmbrOfDaysToChkFor,oneDayMS,parentFldr;

  /* Either pass in the ID of the parent folder to the function or set
   * the parent folder ID in the code below
  */

  nmbrOfDaysToChkFor = 1;//Number of days to check for as being older than

  if (!parentFolderID) {
    parentFolderID = "Enter the folder ID to start the search";
  }

  oneDayMS = 24 * 60 * 60 * 1000;//number of milliseconds in one day

  if (nmbrOfDaysToChkFor > 1) {
    msInTimePeriod = oneDayMS * nmbrOfDaysToChkFor;
  } else {
    msInTimePeriod = oneDayMS;
  }

  d = new Date();

  parentFldr = DriveApp.getFolderById(parentFolderID);

  folders = parentFldr.getFolders();//Get all folders within the parent folder

  while (folders.hasNext()) {//Is there another folder in the folder iterator
    folder = folders.next();
    dateCreated = folder.getDateCreated();

    Logger.log(folder.getName() + " - Date:" + dateCreated);

    diffInMS = d.getTime() - dateCreated.getTime();

    Logger.log('diffInMS: ' + diffInMS)

    if (diffInMS > msInTimePeriod) {
      folder.setTrashed(true);
      Logger.log('folder is trashed: ' + folder.isTrashed());
    }
  }
}
```
And any folders in my Motioneye folder that is older than one day is place into trash and with another AppScript I automatically clean my trash every week

ccrisan, 2019-08-11:
Closed via #1368.

closed 2019-08-11 by ccrisan