motioneye-project/motioneye #1311: Restore from Backup not working?

jordipalet, 2019-06-14:
Hi,

Using Ubuntu.

I did a backup of my configuration (5 cameras), and saved it correctly.

Upgraded with pip install motioneye --upgrade and the cameras are gone.

I restored the backup and restarted and still say no cameras defined ...

even if I restore manually the contents of /etc/motioneye, it looks like not recognizing the config files.

Any hints?

zagrim, 2019-06-14:
You are not telling the versions you upgraded from and to, but just a guess: Have you followed the instructions mentioned in [the release notes](https://github.com/ccrisan/motioneye/releases/tag/0.40)? You need to run configuration migration when upgrading from 0.39 to 0.4x.

jordipalet, 2019-06-14:
Missed that, thanks a lot, just did it and it worked perfectly !

closed 2019-06-15 by ccrisan