motioneye-project/motioneye #1312: Add Bearer token Autorization

martimarkov, 2019-06-14:
I have a camera that uses `Authorization: Bearer {token}` to authenticate requests. 

I think the only file that needs changing on the backend side will be: `motioneye/motioneye/mjpgclient.py`

ccrisan, 2019-06-15:
I don't think Motion supports token-based authentication either.

martimarkov, 2019-06-26:
Does motion eye pass the original video stream location to motion or does it proxy it?

ccrisan, 2019-06-26:
motionEye itself does the proxying stuff but I doubt that's what you're after. Normally clients connect to the MJPEG stream exposed by Motion. Only the motionEye UI uses proxied streaming and downloads frame by frame.

JamBalaya56562, 2026-11-10:
Closing, as this works without a change: for `http://` camera URLs Motion (4.4 and later) opens the stream with FFmpeg and passes `netcam_params` on to it, so FFmpeg's `headers` option can send the token. Add it to the camera under Extra Motion Options:

`netcam_params` = `headers=Authorization: Bearer <token>`

Checked with Motion 4.7.1 against a test server that answers 401 without the token: the header arrives and the camera connects. motionEye keeps the option when the settings are saved, and only appends its own `keepalive` and `tolerant_check`. This does not work for URLs starting with `mjpeg://`, which Motion reads without FFmpeg.

closed 2026-11-10 by JamBalaya56562