motioneye-project/motioneye #1313: Improvement: Display slider value when hovering the cursor

zagrim, 2019-06-15:
One small thing I've been missing for quite a while in motionEye UI is slider value tooltips. In some cases it would really be helpful for the user if sliders showed their current value (as tooltip, just like they do when adjusting the value) when hovering them. One such case is the motion detection threshold setting, as it has a very fine scale and the cursor position doesn't really tell the exact value. The tooltip would also be nice in other similar cases when there are multiple snap positions between displayed scale values.

I can't think of a case where it would be bad to have such a feature implemented in the sliders, so I went and tried that out myself, and will create a PR soon for it.

ccrisan, 2019-06-15:
Yeap, that's something that's been missing from the mobile UI. I'd be interested in seeing how you'd implement it.

zagrim, 2019-06-15:
I don't really know what you're referring to with "mobile UI", I only know of one UI, the web UI, where it's rather easy to do, in contrary to, say, a strictly touch oriented UI where "hover" kinda loses purpose.

zagrim, 2019-06-15:
I did now test the UI on my phone's browser (I guess it was the first time) and while the overal GUI layout works fine on the small screen, the sliders don't work that nice (they don't respond to drag, have to tap on the scale to set new position and of course due to that there's no value tooltip at all).
So, if the UI is aimed to be really mobile-friendly, those sliders would benefit from some work. 
What I did works only when there's a mouse or touchpad involved.

PR #1314 (2019-06-16): Show slider cursor value on mouse hover
Fixes #1313.
closed 2019-06-16 by ccrisan