motioneye-project/motioneye #1316: Streaming password including a colon break camera stream preview

sibbl, 2019-06-15:
I just got the error `too many values to unpack` at this line of code: https://github.com/ccrisan/motioneye/blob/fee12aaefac1e28992795a25c5005d06f0114a33/motioneye/mjpgclient.py#L271

It turns out that the way the string is currently split is not the usual implementation of basic auth strings. Taking HTTP basic auth as the reference, usernames must not include colons, but passwords with colons are allowed. See section 2 of [RFC 7617](https://tools.ietf.org/html/rfc7617).
I'd suggest to use `.split(':', 1)` to follow the standard and avoid the problem.

Furthermore, the streaming basic auth also doesn't follow the standard as the username is not case insensitive. Maybe this can be fixed as well at some point.

ccrisan, 2019-06-21:
Good point. Hoping that Motion actually follows the standard closer than motionEye, I'll try to fix these in motionEye as well.

yztink, 2019-09-06:
I had issue ever since I installed motion eye for the first time last November and just figured out this was the issue yesterday...after many fruitless hours of googling and giving up hope. 🙁 It may have also prevented me from logging in using the surveillance account, but I can't be sure.