motioneye-project/motioneye #1317: Motion detection doesn't work when continus recording mode is enable

sharco62, 2019-06-15:
Hello, 

When I activate the continuous recording function, motion detection no longer works. It remains constantly in detected motion mode. Is this normal?

How can I record continuously and do alerting on motion detection?

Thank you for your help.

ilovelinux, 2019-06-19:
I can't replicate your issue, it works for me. Can someone do it?
What kind of motion detection alert are you using. Web Hook URL just run fine for me.

zagrim, 2019-06-20:
I can replicate this. Having email notification set for Motion Notification and enabling Movies -> Recording Mode : Continuous Recording, all I get is motion notifications every <max movie length> seconds, and indeed the camera stream shows red frames as if motion was continuously detected. Looking at the code this seems obvious: "continuous recording" just enables "emulate_motion" on camera config, which basically means that motion pretends that it is detects motion all the time.

I guess you just can't have both at the same time, continuous recording and motion detection of any kind.

thrust15, 2019-08-02:
I have the same issue. I want continuous recording and an event when motion starts and ends (to publish the state to home assistant). The end command never occurs.

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #736: continuous recording works by making Motion treat every frame as motion (`emulate_motion`), as @zagrim found here and @ccrisan explained there, so motion events and their notifications cannot be told apart from it today.

closed 2026-11-13 by JamBalaya56562