motioneye-project/motioneye #1318: Request: Is there a way to not show camera feeds until login?

mikepruett3, 2019-06-18:
I don't know if this is possible functionality or not... but I would like a way to turn off camera streams at the default page... until login.

Currently, if you browse to the MotionEye page, all camera streams are displayed (but not controllable). I would like to provide access to my MotionEye install remotely, but only to users whom have logged in only.

Is there a fix for this? Having public access (visable) to RTSP streams of my cameras just feels creepy!

CRImier, 2019-06-18:
That's weird. I'm not the main developer, but I manage a system of many interconnected MotionEye cameras and I don't have this issue - when opening the motioneye webpage, I see the login page without any feeds (just "enter your username and password"), unless I've logged in using that browser before, of course. Do you have both user and admin passwords set (non-empty)? Are the feeds password-protected (uner the "video streaming" tab? [unrelated, but make sure they are] Can you maybe provide a screenshot?

mikepruett3, 2019-06-18:
Thank you! That was it! I only had the admin password filled out, not the user password.

CRImier, 2019-06-18:
great! Since your problem is solved, please close this issue =)

closed 2019-06-18 by mikepruett3