motioneye-project/motioneye #1319: [FEATURE REQUEST] Custom settings based based on ip range

ilovelinux, 2019-06-18:
When I'm at home and my device is connected on the local network (192.168.x.y with y != router ip), I can stream at maximum quality without any buffering/frame skipping issue. When I'm outside and access to MotionEye by public ip (192.168.x.y where y is the router's ip), my upload speed is a bottleneck so I would to lower the quality/framerate in order to have a still decent streaming. When I reach MotionEye through Tor network (127.0.0.1), speed will be even lower than when I connect through public ip so I want to customize my settings even more in order to have a decent stream even through Tor.

jwheeler91, 2019-06-25:
Sounds like a nice feature, but I fear its asking a bit much from a free open source project...

A work around for you would be to use the admin account with framerate and resolution dimmer preferences at maximum and the normal user with preferences at 50%. But you will have to log out and log in when you take a device away from home.

ccrisan, 2019-06-26:
This is something very specific and has nothing to do with motionEye. You'll need to set that up using other tools available on your OS, such as iptables firewalls and/or http reverse proxying using e.g. Nginx.

closed 2019-06-26 by ccrisan

ilovelinux, 2019-06-26:
I asked settings per-user or per-user-category. It can't be set from firewall or proxies, it must be handled by the server backend which manage logins

jwheeler91, 2019-06-27:
To be fair, you did ask two questions in one and I can see (based on the last question) why Calin suggested using nginx to block external access, missing the bit about only blocking the admin user.

Something I wish existed was a way to have the admin version of the interface only available via a '/admin' url because then you could control access using location blocks in nginx using reverse proxy. (I would actually then remove my admin password lol... as it would be controlled by the nginx auth module)....  P.s. I already block access to my cameras except over the LAN inc VPN, but with a separated admin location, you could do this for just the admin interface...

I have thought of another workaround too:
- Two docker containers running motioneye (I made my own based on linuxserver.io's architecture)
- One is your main one running at full resolution, doing your motion detection, recording etc...
- The second is a "viewer" only, you would add the cameras from the first docker rather than point to the actual cameras using the camera type "remote motioneye camera"
- Set your framerate and resolution dimmer preferences to 50% on the external one...
- I found that "remote motioneye camera" meant that I was controlling the remote camera rather than the local copy, but that might be because I used the admin user...?
- Run nginx (I reccomend using docker) to reverse proxy to motioneye main if local, and motioneye remote if remote