motioneye-project/motioneye #1320: Fix installation guide for Archlinux

ilovelinux, 2019-06-18:
Motion isn't on official Archlinux repositories but in AUR. Arch users need to follow this steps:

1. Install base-devel and git: `$ pacman -S base-devel git`
2. Clone Archlinux's aur package: `$ git clone https://aur.archlinux.org/motion.git`
3. Join in the directory and build: `$ cd motion && makepkg -is`

ccrisan, 2019-06-21:
I had no idea they removed it from official repos. I've now update the wiki page. Thanks.

closed 2019-06-21 by ccrisan

ilovelinux, 2019-07-07:
They just re-added it to the official repos: https://www.archlinux.org/packages/community/x86_64/motion/ (Also in ARM repos, just checked)

ilovelinux, 2020-02-22:
@ccrisan as I said you should remove the commands to build Motion by themselves because it has been added to the official repository again

ccrisan, 2020-03-09:
Updated. Thanks!