motioneye-project/motioneye #1321: Encoding error about webhook request url

ilovelinux, 2019-06-19:
Webhook URL (GET): `https://api.telegram.org/bot123456:ABC-DEF1234ghIkl-zyx57W2v1u123ew11/sendMessage?chat_id=123456789&text=Qualcosa+si+è+mosso`
Here's the log:
```
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
   ERROR: 'ascii' codec can't encode character u'\xe8' in position 240: ordinal not in range(128)
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 273, in get
    self.list()
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 660, in list
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1408, in motion_camera_dict_to_ui
    e = shlex.split(e)
  File "/usr/lib/python2.7/shlex.py", line 275, in split
    lex = shlex(s, posix=posix)
  File "/usr/lib/python2.7/shlex.py", line 25, in __init__
    instream = StringIO(instream)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe8' in position 240: ordinal not in range(128)
   ERROR: 500 GET /config/list/?_=1560961996973&_username=admin&_signature=93c08708f088e5f5a6477aeb16e9b481831cdf0d (192.168.188.32) 45.06ms
   ERROR: 'ascii' codec can't encode character u'\xe8' in position 240: ordinal not in range(128)
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 270, in get
    self.get_config(camera_id)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 317, in get_config
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1408, in motion_camera_dict_to_ui
    e = shlex.split(e)
  File "/usr/lib/python2.7/shlex.py", line 275, in split
    lex = shlex(s, posix=posix)
  File "/usr/lib/python2.7/shlex.py", line 25, in __init__
    instream = StringIO(instream)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe8' in position 240: ordinal not in range(128)
   ERROR: 500 GET /config/1/get/?force=true&_=1560961996981&_username=admin&_signature=a2337eadfe21b78636c265d0944a1bbb89431e4a (192.168.188.32) 17.60ms
 WARNING: 403 GET /login/?_=1560962003209 (192.168.188.32) 6.49ms
   ERROR: 'ascii' codec can't encode character u'\xe8' in position 240: ordinal not in range(128)
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 273, in get
    self.list()
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 660, in list
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1408, in motion_camera_dict_to_ui
    e = shlex.split(e)
  File "/usr/lib/python2.7/shlex.py", line 275, in split
    lex = shlex(s, posix=posix)
  File "/usr/lib/python2.7/shlex.py", line 25, in __init__
    instream = StringIO(instream)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe8' in position 240: ordinal not in range(128)
   ERROR: 500 GET /config/list/?_=1560962008259&_username=admin&_signature=be254dd134607f51dfb7279ad09d09fa98c38ce8 (192.168.188.32) 17.14ms
```
There are two workaround:
- Change the OS locale to an UTF-8 one (System side)
or
- Encode the text before saving it with [urllib.urlencode](http://docs.python.org/2/library/urllib.html#urllib.urlencode) (Motioneye side, fix it)

Kwickell, 2019-07-03:
Getting the same type of errors.
Any updates on this?
Similar error log as @ilovelinux.

Is there anything else i could do to help with troubleshooting this error?

`2019-07-03 00:13:10: [motioneye]    ERROR: 'ascii' codec can't encode character u'\xe5' in position 15: ordinal not in range(128)
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1499, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1700, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 290, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 530, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 371, in set_camera_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 449, in set_camera
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 242, in set_main
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 2072, in _set_additional_config
  File "/usr/lib/python2.7/site-packages/motioneye/wifictl.py", line 203, in _set_wifi_settings
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe5' in position 15: ordinal not in range(128)
2019-07-03 00:13:10: [motioneye]    ERROR: 500 POST /config/1/set/?_username=admin&_admin=true&_signature=2984d7004232bf52c80f7f8b9856f53a5a4d2fb1 (192.168.86.101) 103.09ms`

zagrim, 2019-07-07:
I'm working on a fix for this.

zagrim, 2019-07-09:
@Kwickell I'm not absolutely sure about this, but do you have non-ascii characters in your Wifi config (SSID or passcode)?

Kwickell, 2019-07-09:
@zagrim Yeah, sadly the name contains an "å"; passwords only strange char is a space.

Didn't see the details being transferred in the POST. Could that possibly be related to another issue I've been seeing? After reboot the OS is stuck in a reboot-loop because network is having critical errors.

zagrim, 2019-07-09:
I have no experience on the Wifi control feature, and I don't have a setup where I could/want to try it myself in practice, but I can try to set a dummy wifi config on my box and see how it behaves. It should be easy enough to deal with character encoding related issues, I hope.

Kwickell, 2019-07-09:
@zagrim Yeah it might actually be an underlying issue in the OS.
I've had trouble with it before but setting the correct settings in the OS solved it.

Might have to make the rounds and replace the Wifi SSID here at home ;)
I'll do that and update here if that fixes it.

zagrim, 2019-07-11:
@Kwickell After having looked a bit at it, I guess wifi control is only enabled in motionEyeOS and not available at all in the standalone application (which kinda makes sense). Thus I can't really tackle that now. I hope eliminating non-ascii chars from SSID helped.

Kwickell, 2019-07-11:
Seems like the change of SSID did the job. Still have a few errors similar to this in a Hass.io integration.My issues seems solved. Thanks for the assist! 😊

PR #1362 (2019-07-27): Work around deficiencies with unicode in shlex
Fixes #1321. It seems shlex is not good with unicode. I'm not quite sure if this would not be necessary at all if we were on Python 3.

I noticed that non-ascii in root dir or image file names is not that good, either, but I would take another approach with that so I left it out of this PR.
closed 2019-07-27 by ccrisan