motioneye-project/motioneye #1322: Don't show APPLY button if nothing has been changed

ilovelinux, 2019-06-19:
APPLY button appears even when nothing has been really changed. It's enough to click a slider, add a space/delete and rewrite text on textboxes and so on.

jwheeler91, 2019-06-25:
This would be a nice "nice to have" feature. A few methods spring to mind:

- Perform a diff on the whole config for every change to see if there are any changes. But this will increase the amount of client side processing, potentially slowing down weaker devices.
- Store the "original value" in each field's html tag. Then the js only needs to compare the one value that changed. But this *could* increase network traffic. (download size) A mitigation for this would be to add the original values to the fields when a change occurs, thereby keeping network traffic to a minimum.

I should note, that I am presuming that "onclick" of any setting will "un-hide" the apply button.

ilovelinux, 2019-06-26:
Could onfocus register the value to compare when the field is unfocused?

jwheeler91, 2019-06-27:
Yes, so on focus to store, potentially with a class of 'clicked' or something to help with select all elements that have been changed. However I would use whatever the current method uses to show the apply button to do the comparison to minimise the change and minimise the unexpected outcome...

I don't know how the on focus events work with the sliders, but I guess using classes can help a lot with reusable code?

Either way, this would be a very low priority feature if this was my project. Mainly because you would likely be setting and forgetting the settings (after tuning). :)

zagrim, 2019-06-27:
As a user I have been wondering if there should be a better way to keep track of changes since the apply button appears maybe a bit too often. After getting slider values as tooltip (currently in dev version only, I think) it's not so pressing, perhaps (no reason anymore to click on a slider to see the current exact value).

Anyway, I peeked at the code and unless I am mistaken, the Apply button is made visible in the overall settings change event handler for every change (as @jwheeler91 presumed), so to me it seems there's not much to build on.