motioneye-project/motioneye #1323: [FEATURE REQUEST/SECURITY ISSUE] Allow to change admin username

ilovelinux, 2019-06-19:
If everyone knows that admin username can't be changed, it will be easy for a malicious user to brute force the password specially in the local network.

jwheeler91, 2019-06-25:
I use nginx with an auth module (that use LDAP) protecting the cameras... There are even simpler auth methods though...

ccrisan, 2019-06-26:
Having to guess two fields instead of one doesn't practically increase the security of a system. You could simply increase the length of the password to make it more difficult for the attacker to guess it.

closed 2019-06-26 by ccrisan

ilovelinux, 2019-06-26:
It does. If hacker must guess both fields, O(n) becomes O(n²) and it's not irrelevant.

CRImier, 2019-06-27:
You might want to, i.e., double your password length if you're that worried about the amount of effort the hacker woud need to expend. Or, i.e., insert a simple proxy that asks for as many logins&passwords as you want, maybe also does some other access control. There are good ways to protect devices on your local network from an attacker that's inside, this doesn't seem like a good way to do it (in my experience managing a small network, at least).

Besides, unless you have TLS enabled in your local network, aren't the logins&passwords sniffable anyway while you're logging in?