motioneye-project/motioneye #1324: Video not showing when chrome lazyload feature is enabled

rajendrant, 2019-06-20:
When chrome lazyload feature is enabled, motioneye videos are not showing up.
This is caused by motioneye javascript code using property name 'loading' for img elements.
This property name is reserved when lazyload feature is enabled, and breaks motioneye setups.

See this chromium bug for more details
https://bugs.chromium.org/p/chromium/issues/detail?id=976468#c2

bob-lee, 2019-06-22:
I wonder `loading` property name is reserved by whom / when..

rajendrant, 2019-06-26:
The property name is being reserved by the html standardization group
https://github.com/whatwg/html

So all browsers would support it.
Though the name clash bug would essentially only trigger when they fully implement it.

closed 2019-07-01 by rajendrant