motioneye-project/motioneye #1327: How to get always the same Image File Name?

bojanopatje, 2019-06-21:
Hi.
i'm making webcam on my personal website that will show some landscape, but I need that motioneye upload on the server, file with always the same name "picture.jpg". On motioneye settings it appears that I can't do this (%Y-%m-%d/%H-%M-%S). Is there anny other way?
Thank you.

michalchrzastek, 2019-06-24:
Of course you can do it, just overwrite the file name.
Then in the file browser there is only 1 photo and is overwritten every second or every frame. 

[image]
[image]

bojanopatje, 2019-06-25:
Thank you. I asked before I tried. I saw that it is possible to name by date, time,... and I thought that it is not possible in other way. Thank you again.

closed 2019-06-26 by ccrisan