motioneye-project/motioneye #1329: Insert a camera-stream from another motioneye

holgerbusch, 2019-06-24:
Hello from Germany.
I have one raspberry with a Pi-Camera with motioneye running. 
Now I want to run a second raspberry with motioneye as well - like a server, so that the first one just used like an IP camera. 
I activated the streaming with no authentication on raspberry one and I can watch it by opening the stream in a browser. 
When I want to insert an camera into the second, motioneye finds the stream and insert it, but theres no picture to see. 
I installed both raspberries with fresh downloaded motioneye yesterday. 
Maybe you can help me to solve the problem?

Thanks from Germany.
Holger
[image]

michalchrzastek, 2019-06-24:
I had this issue as well. Set the first camera as FastNetworkCamera under Expert setting, then on your second Pi, add it again. Try those settings (obviously with your own IP).

[image]

holgerbusch, 2019-06-24:
@michalchrzastek thanks.
That worked.

Amazing.
Thanks from Germany.

ilovelinux, 2019-06-26:
Problem has been solved. Please @holgerbusch close this issue

closed 2019-06-27 by holgerbusch