motioneye-project/motioneye #1330: Wifi Off?

Henrietta1989, 2019-06-24:
Is it possible to have 1 USB-camera in Raspberry and after setup, switch WiFi off and cameras continue work after that?

ilovelinux, 2019-06-26:
Obviously yes. Motioneye will keep working and running

Henrietta1989, 2019-06-27:
I switched WiFi OFF and camera feed simply disappeared. That makes it unusable to me.

So what's wrong. Thanks

zagrim, 2019-06-27:
@Henrietta1989 what's your setup regading motionEye like? Do you access the motionEye user interface directly on the Raspberry (keyboard, mouse and display connected to the Raspberry), or do you access motionEye on a browser from a laptop or something like that? 
If it is the latter, do you have any other network connecting the Raspberry to the laptop except the Wifi?

Henrietta1989, 2019-06-27:
I just installed latest NOOBS 3.1.1 Raspbian and then installed latest MotionEye and i access it directly on the Raspberry. It just have latest Raspbian and MotionEye. Thanks

ilovelinux, 2019-06-27:
You are probably connecting to Motioneye with your local IP address which is given to you from the router (192.168.x.y). You should instead connect to Motioneye through localhost (127.0.0.1)

JamBalaya56562, 2026-11-10:
As answered above: when the browser runs on the Pi itself, open `http://127.0.0.1:8765` (or `localhost`) instead of the address the router gave the Pi, and the cameras keep working with Wi-Fi off. Closing as answered.

closed 2026-11-10 by JamBalaya56562