motioneye-project/motioneye #1331: first 10sec of movie blank screen

michalchrzastek, 2019-06-24:
Hi,

Thank you for developing and looking after this project!

I have MotionEyeOS installed on my RasPi2 and RasPiZero. The Zero is set to Fast Network Camera and Pi2 is set to continuously record movie. I've set Max Movie Length to 300 seconds. When reviewing the footage first 10 seconds of every movie is just black screen. 

Changing the Max Movie Length to any value doesn't make a difference. I don't want to have it set at 0 in case disk or file gets corrupted so I loose all footage.

I have disabled Pi0 as Fast Network Camera, and enabled movie recording to test, but had the same result, first 10 sec is missing.

I'm running motionEyeOS 20181209.
Any tips?
Thanks.

jasaw, 2019-06-25:
What value did you set for your "pre-capture frames" ?

michalchrzastek, 2019-06-26:
Do you refer to "Captured Before" setting under MOTION DETECTION? If so this is set to 1, anyway motion detection is disabled.

JamBalaya56562, 2026-11-08:
Closing, as there were no further replies, and this was on motionEyeOS 20181209 with continuous recording. The movies are encoded by Motion, so a black start in every movie points to Motion or the camera, not to motionEye. If movies still start black with current releases, please report it to [Motion](https://github.com/Motion-Project/motion), or to motioneye-project/motioneyeos for motionEyeOS.

closed 2026-11-08 by JamBalaya56562