motioneye-project/motioneye #1332: Feature to upload a snapshot as soon as movement is detected

rodrigouroz, 2019-06-24:
I want to fork the repo and create a new feature for uploading a snapshot (the first frame perhaps) when movement is detected. 
The idea is to upload a picture as quickly as possible when movement is detected, while the clip is being recorded and uploaded.
I tested MotionEye and loved it, however when movement is detected you have to wait for the clip to be finished and uploaded, I'm thinking on a situation where the hardware running this is tampered or disabled before the clip is uploaded. Uploading a picture should be faster.
I'm willing to implementing this myself and wanted to ask recommendations to people more used to this repo. Where should I start looking?

ccrisan, 2019-06-26:
I don't think you need to fork anything: you can use the "Run A Command" option to execute a command whenever motion is detected. That corresponds to `on_event_start` Motion option.

rodrigouroz, 2019-06-26:
Ok, it's not clear to me which info I'll get in the command from that event by looking at the code, I'll give it a try. I guess I could grab a frame from the detected camera as soon as the command is called.

ccrisan, 2019-06-26:
@rodrigouroz you might need to look at [Motion](https://github.com/Motion-Project/motion) as well. Grabbing the first frame might be more difficult than it sounds.

JamBalaya56562, 2026-11-10:
No fork is needed for this. Set Still Images → Capture Mode to "Motion Triggered (One Picture)" and turn on "Upload Pictures" under Upload Media Files: the picture is saved when motion starts and uploaded right away, while the movie is still being recorded. Closing as answered.

closed 2026-11-10 by JamBalaya56562