motioneye-project/motioneye #1335: Ignoring libssh2 on Raspbian

ovidiub13, 2019-06-25:
I've installed everything exactly as the wiki steps say, but when starting the service it fails with this error:

```
Jun 26 01:15:56 raspberrypi systemd[1]: Started motionEye Server.
^[[BJun 26 01:15:59 raspberrypi meyectl[6554]: Traceback (most recent call last):
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/bin/meyectl", line 11, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     sys.exit(main())
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import server
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 31, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import handlers
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 30, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import config
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 32, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import motionctl
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 28, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import mediafiles
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 41, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import uploadservices
Jun 26 01:15:59 raspberrypi meyectl[6554]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 27, in <module>
Jun 26 01:15:59 raspberrypi meyectl[6554]:     import pycurl
Jun 26 01:15:59 raspberrypi meyectl[6554]: ImportError: libssh2.so.1: cannot open shared object file: No such file or directory
Jun 26 01:15:59 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jun 26 01:15:59 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
Jun 26 01:15:59 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

Searching for the library I get the following:

```
pi@raspberrypi:~ $ ldconfig -p | grep -i libssh
        libssh_threads.so.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh_threads.so.4
        libssh_threads.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh_threads.so
        libssh.so.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh.so.4
        libssh.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh.so
        libssh-gcrypt_threads.so.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh-gcrypt_threads.so.4
        libssh-gcrypt.so.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh-gcrypt.so.4
pi@raspberrypi:~ $ ls -lah /usr/lib/arm-linux-gnueabihf/libssh*
-rw-r--r-- 1 root root 205K Apr  5 00:32 /usr/lib/arm-linux-gnueabihf/libssh2.a
lrwxrwxrwx 1 root root   16 Apr  5 00:32 /usr/lib/arm-linux-gnueabihf/libssh2.so -> libssh2.so.1.0.1
lrwxrwxrwx 1 root root   16 Apr  5 00:32 /usr/lib/arm-linux-gnueabihf/libssh2.so.1 -> libssh2.so.1.0.1
-rw-r--r-- 1 root root 150K Apr  5 00:32 /usr/lib/arm-linux-gnueabihf/libssh2.so.1.0.1
-rw-r--r-- 1 root root 587K Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh.a
lrwxrwxrwx 1 root root   22 Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh-gcrypt.so.4 -> libssh-gcrypt.so.4.4.1
-rw-r--r-- 1 root root 379K Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh-gcrypt.so.4.4.1
lrwxrwxrwx 1 root root   30 Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh-gcrypt_threads.so.4 -> libssh-gcrypt_threads.so.4.4.1
-rw-r--r-- 1 root root 5.4K Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh-gcrypt_threads.so.4.4.1
lrwxrwxrwx 1 root root   11 Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh.so -> libssh.so.4
lrwxrwxrwx 1 root root   15 Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh.so.4 -> libssh.so.4.4.1
-rw-r--r-- 1 root root 379K Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh.so.4.4.1
-rw-r--r-- 1 root root 1.8K Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh_threads.a
lrwxrwxrwx 1 root root   19 Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh_threads.so -> libssh_threads.so.4
lrwxrwxrwx 1 root root   23 Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh_threads.so.4 -> libssh_threads.so.4.4.1
-rw-r--r-- 1 root root 5.4K Dec 31 15:47 /usr/lib/arm-linux-gnueabihf/libssh_threads.so.4.4.1
```

So for some reason `ldconfig` is ignoring libssh2.

Any ideas?

ccrisan, 2019-06-26:
Is this a recent Raspbian? While this is definitely not a valid motionEye issue, we could try to investigate and see what's the root cause of this problem. 

My Raspbians have only `libssh2` and no `libssh`. My `ldconfig -p | grep ssh` says:

    libssh2.so.1 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libssh2.so.1

ovidiub13, 2019-06-26:
I did the following:
* `apt-file` to find the packages for the other `libssh` libs, and removed them
* `apt autoclean && apt autoremove`
* `apt-file` to find the package containing `libssh2`
* force the reinstall of it `apt --reinstall install libssh2-1`

And it's working now. Thank you.

closed 2019-06-26 by ovidiub13