motioneye-project/motioneye #1337: More settings?

Henrietta1989, 2019-06-27:
Are there any more settings i could use to customize MotionEye even more?

Advanced settings is on and i can see all "Preferences ... Working Schedule" settings.

Any plugins?

ccrisan, 2019-06-27:
Nope. No plugins.

closed 2019-06-27 by ccrisan

jurgenweber, 2020-02-25:
how do you turn on advanced settings? I can find no such option

ccrisan, 2020-02-25:
@jurgenweber you don't. Advanced Settings are gone. Well, not the settings themselves, they are all visible now to admin users.

jurgenweber, 2020-02-25:
ah, cool. A little confusing since there are references to them in the doco.

ccrisan, 2020-02-25:
@jurgenweber please point them out so we can remove then.

jurgenweber, 2020-02-25:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#image-from-docker-hub

4th paragraph down.