motioneye-project/motioneye #1342: Incorrect help text for smart mask sluggishness

zagrim, 2019-06-30:
motionEye GUI says about Smart Mask Sluggishness:
> lower values result in a longer-lasting smart mask with a slower building process

which is on the first sight in line with what motion docs say about smart_mask_speed:
> 1 is slow, 10 is fast

but in config.py the values are actually inverted:
`ui['smart_mask_sluggishness'] = 10 - data['smart_mask_speed']`

So, I guess the help text in main.html needs to be changed?

I have not actually verified this with the functionality as I don't have a suitable setup handy (not a windy day today :smiley:  ) so please forgive me if I understood something wrong.

zagrim, 2019-06-30:
Seems also like setting sluggishness to max (10) causes actually smart mask to be disabled (since the value converts to 0 in config file, which means smart mask is disabled).
This makes me wonder if the meaning of the option value has been altered in some earlier version of motion, and even if the current implementation might have originally been good it got outdated at some point?

ccrisan, 2019-08-23:
@zagrim you're right. The description was wrong. Fixed via https://github.com/ccrisan/motioneye/commit/be999e2574e1b3e8c0f9e7aeb313cec51ee3703d.

closed 2019-08-23 by ccrisan

lutierigb, 2020-01-21:
#973