motioneye-project/motioneye #1343: Video not showing in motioneye frontend

martinschm, 2019-06-30:
Hi,

I am running 
`motionEye Version | 0.40
-- | --
Motion Version | 4.0
OS Version | Debian 9.9
`
I have attached two Xiaomi Xiaofang S1 cams via rstp over wifi (UniFi AC Pros) that run the danfang hack. While I was on vacation it was all running smoothly and I was able to pull images to my smart home system and send them via Telegram.

Yesterday night I logged into the motioneye frontend and it looked like this
[image]

Both cams are operational and the video server/rstp stream works. However motioneye does not show a picture. Is this an issue related to motioneye or the system it runs on or is it something related to the client computer looking at the frontend ?

motioneye runs on a Rock64 (4GB Ram) that is connected via ethernet to my router.

Best
Martin

objectiveinteraction, 2019-07-01:
I confirm this is true as well. 

Right so, the main issue is really a question of how things are named. There is motioneye and there is motion. Once motioneye is started, it tries to start motion the server in the background to provide feeds to the motioneye web frontend. Motion itself also has http endpoints that can be used as well as a web interface.  The documentation doesn't quite make that clear, and the user is left wondering if options in the configuration files are meant for motion or motioneye.

So the general idea is to start motion first without using motioneye and see if it works. I had to add configuration details to start the web interface of motion. From there you can see the streaming videos, so motion itself is working. 

Switching to motioneye, starting it on the command line, i get a lot of "WARNING: Connect error on fd XX: ECONNREFUSED". This string format exists only as a motion warning string. And from the motioneye interface, i cannot see the image. So, it has to be some kind of configuration error by motioneye while calling the motion server to start.

objectiveinteraction, 2019-07-01:
The error string also seems to be a connection refused error warning in  python and grepping through the motioneye codebase, it seems the mjpgclient.py could be the reason why

martinschm, 2019-07-01:
Hi,
where do you see this error popping up ? 
I want to double check whether I have the same error happening.

The odd thing is that I did not change anything. While I was on vacation everything seemed to have worked. Once I was home it stopped working. 

Now thinking about it. I did a apt-get update before installing some other software. I guess that could caused some packages used by motioneye to have been updated and hence trigger the running system to break.

martinschm, 2019-07-01:
Funny observation. I turned off the second cam and suddenly the first one was showing up in motioneye again. Turning on the second cam, both do not display anything.

I had the same thing before my vacation when trying to turn on a third cam.

Can this be related to network throughput on my host system ?

luizbird, 2019-08-15:
Hello.

Any news on this? I'm having the same issue but I have only one camera configured.. it stopped working 3 days ago.. without any reason or changes on system (at least, not from my side).

CRImier, 2019-08-15:
Hi! Any chance any of you are using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

luizbird, 2019-08-15:
Hi there! Thank you for the fast feedback.

I run it on Docker, so I will check which version I'm running. Neverteless, the chrome trick worked! Thank you.

ccrisan, 2019-08-19:
Problem fixed in 0.41rc1.

closed 2019-08-19 by ccrisan

carmenchui, 2020-09-03:
Hi @ccrisan, I'm also getting this error on both Chrome 85.0.4183.83 and Firefox 78.2.0esr on the same setup with two Xiaomi Xiaofangs.  The lazy-image flag is enabled on both browsers (per https://github.com/ccrisan/motioneye/issues/1401), but no image appears on the motioneye front-end.  I've checked that the RTSP feeds work on VLC media player.

goxofy, 2022-10-22:
same problem