motioneye-project/motioneye #1344: Re-connected but still "enabled" video devices require manual intervention to show in the MotionEye dashboard

PieBru, 2019-06-30:
Hi Calin,

I am temporary using some old D-Link DSC933 wifi cams, while I progressively substitute all my cams with MotionEyeOS things.
All my commercial cams occasionally disconnects from wifi. When they reconnects, ME doen't show them on the dashboard, even after one hour or more. This happen also with market cams other than D-Link, so I think this issue may affect also other users.

IMO it should be nice to periodically "ping" the cams marked off-line by ME, to recognize if and when they become online again. This could help also to mitigate some DHCP renew issues.

Meanwhile, when such crappy cams disconnect/reconnect, to re-enable them in the ME dash, for each off-line cam I manually disable (OFF) the video device, "apply" to save the new config, enable (ON) and "apply" again on it. I must do this for each "re-connected" cam, then I see it almost immediately on the ME dash.

This is my ME docker install under hassio:
motionEye Version | 0.40
Motion Version | 4.2.2
OS Version | Linux 5.1.15-arch1-1-ARCH

Thanks,
Piero

ccrisan, 2019-07-01:
Does the underlying Motion daemon reconnect to these cameras? You can check that by trying to connect to the corresponding MJPEG streaming port (808x, where x is the camera number, starting from 1). If it doesn't, this sounds more like a [Motion](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) issue.

PieBru, 2019-07-02:
When this happens, the camera reappears on the MotionEye dash few seconds after accessing the MotionEye camera stream with the web browser.
Btw, here the stream is HTTP, not HTTPS as suggested in the MotionEye link.
Piero

JamBalaya56562, 2026-11-08:
Closing, as this was with the motionEye and Motion of 2019. As ccrisan pointed out, reconnecting to a camera is done by Motion; your note that the camera came back once its stream was opened points there too. The web UI itself keeps trying a camera that failed, once per second. If a reconnected camera still stays offline with current releases, please report it to [Motion](https://github.com/Motion-Project/motion).

closed 2026-11-08 by JamBalaya56562