motioneye-project/motioneye #1345: Update Docker build instructions

PetricaM, 2019-07-01:
Hi,

Current instructions for Docker mention exposing only web access port from the Docker, not the streaming ports (8081, 8082, etc):

`-p 8765:8765 8081:8081 8082:8082 \`

kleini, 2019-07-02:
Sorry, streaming ports are mentioned:

> Add additional port mappings with the -p parameter if you want to use the streaming feature of motion.

PetricaM, 2019-07-02:
> You still need to read the text and not just blindly copy the example.

Sarcasm not really needed: I was only trying to improve the documentation if someone else reads it as there is no reference in the page to ports used for streaming that should be passed to the container (ie. 8081 for first camera, 8082 for second, etc)

Anyway, thanks for the great software :+1

kleini, 2019-07-02:
@PetricaM Please make a concrete proposal, how the wiki page should look like or how the documentation would have helped you. This makes it much easier for the maintainers bombed by this or that does not work and this or that is shitty.

PetricaM, 2019-07-02:
Done (only added comments; nothing changed below the line you referenced above so that it can be pasted if agreed upon). Also added some comments regarding the use of higher ports case there are conflicts with other services (but these apply to Docker in general, not only to motionEye).

Add additional port mappings with the `-p` parameter if you want to use the streaming feature of motion: `-p 8765:8765 8081:8081 \` ; for cameras added, numbering of ports used for streaming starts from 8081 (second camera will use port 8082, etc). Ports used for streaming can be later changed in motionEye (Advanced Settings -> Video Streaming -> Streaming Port) but should always match the ones that are being exposed from Docker.

If using additional services that make use of ports in the 808x range, then default mapping can be edited to avoid conflicting by mapping higher range port numbers that are not in use by other services (i.e., `-p 8765:8765 58081:8081 58082:8082 \`)

kleini, 2019-07-03:
I added this to the wiki page. Many thanks for your contribution, @PetricaM .

closed 2019-07-03 by kleini