motioneye-project/motioneye #1346: Cannot run application due to CURL_OPENSSL_3 not found

FredMa01, 2019-07-01:
Hi, 
I upgrade Raspbian stretch to buster.
All is ok but not for motioneye which doesn't restart.
I have this error :
```
import motionctl
meyectl[17196]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 28, in <module>
meyectl[17196]:     import mediafiles
meyectl[17196]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 41, in <module>
meyectl[17196]:     import uploadservices
meyectl[17196]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 27, in <module>
meyectl[17196]:     import pycurl
meyectl[17196]: ImportError: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/lib/python2.7/dist-p
systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: motioneye.service: Failed with result 'exit-code'.

```
I tried to clean cache : 'rm -rf ~/.cache/pip'
No effect !
I tried to upgrade : 'pip install motioneye --upgrade'
No effect !
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: motioneye in /usr/local/lib/python2.7/dist-packages (0.40)
Requirement already satisfied, skipping upgrade: tornado<6,>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.0.1)
Requirement already satisfied, skipping upgrade: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.0.0)
Requirement already satisfied, skipping upgrade: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye) (2.10)
Requirement already satisfied, skipping upgrade: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye) (7.43.0.1)
Requirement already satisfied, skipping upgrade: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (0.5)
Requirement already satisfied, skipping upgrade: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.4.0.3)
Requirement already satisfied, skipping upgrade: futures in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.2.0)
Requirement already satisfied, skipping upgrade: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye) (1.0)
Requirement already satisfied, skipping upgrade: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.11.0)

```
An other idea ?
Fred

ccrisan, 2019-07-01:
The broken package in your case is `pycurl`. Make sure it's properly installed. You could uninstall it using

    `pip uninstall pycurl`

And then reinstall it using

    `pip install pycurl`.

FredMa01, 2019-07-01:
All thanks Ccrisan !
There is no error now but motioneye write "You have not configured any camera yet. Click here to add one.." and doesn't seem to see my thread-1.conf, thread-2.conf, thread-3.conf.
I am looking for why ;)
FRed

closed 2019-07-01 by FredMa01

BennyS91, 2019-07-18:
Did you found out, why the config is not used?

FredMa01, 2019-07-18:
Yes I found. 
I changed in /etc/motioneye/motion.conf the name "thread"
thread thread-1.conf
thread thread-3.conf
thread thread-2.conf

by 
camera thread-1.conf
camera thread-2.conf
camera thread-3.conf

zagrim, 2019-07-19:
Was this a fresh install of motionEye, or upgrade? From your first comment I take it to have been an upgrade, and it sounds like you missed [the release notes for 0.40](https://github.com/ccrisan/motioneye/releases) regarding the need for running the migration script. There might be other config options beside the camera references that should be migrated, too.

FredMa01, 2019-07-19:
It was an upgrade from raspbian stretch to buster.
I haven't seen the release notes for 0.40.
Thanks

fgr1986, 2020-05-14:
You can remove the folder and old lib with
`sudo mv /usr/local/lib/python2.7/dist-packages/pycurl* .
`

and then install it with 
`sudo apt install python-pycurl`

Johnny-quattro, 2020-10-07:
fgr1986, thanks for your post. It helped enormously, as I was having issues after an installation on a Pi that had been upgraded from Stretch to Buster. Cheers.