motioneye-project/motioneye #1347: Run a Command - nothing happens

RobinNowak, 2019-07-01:
Hello,
Im trying to execute a script after a motion is detected in motioneyeos (on a Raspberry Pi)

Im pretty new to all of this so i'll try to give as much Information as possible:

Motioneyeos is installed mainly following this turorial:
https://klenzel.de/3689

Everything is running fine and the function "Send an E-Mail"  also works perfectly after detecting motion.

Now i have a bash script (test.sh) which opens a browser for 20 seconds.

I made it executable wit the command chmod +x 
and it is working after typing "./test.sh " in the termial.
The test.sh file is stored in /home/p 

My test.sh file:
#!/bin/bash
openbox &
openbox_pid=$!
/usr/bin/chromium-browser &
chrome_pid=$!
sleep 20
kill "$chrome_pid" "$openbox_pid"

In the "Run a Command" line i tried things like:

/home/pi/test.sh
bash /home/pi/test.sh
...
but nothing happend so far.

I already did a lot of research and read all of the other issues that were close to my Problem, but i still could not apply the solutions on my project.
I hope you can help me out!

serignembow, 2019-07-11:
Do you still have the issue?
I am trying to run a shell script with motion but I got the same thing.

Pi-DD, 2019-07-11:
I have made this work by storing the script thus ... /etc/motioneye/mycript.sh

and then entering that full directory and script name string in the option in the web interface and applying the change. For me that works for both “run a command” and “run an end command”.

My script initiates an IFTTT webhook.

serignembow, 2019-07-11:
Thanks.
I also changed bash script directory and its okay too.

hulleyrob, 2019-08-04:
Yeah I'm having a problem with the end command seemingly starting but never finishing which I have got round by checking for modified files in a Run a Command script that waits.

aredon, 2019-09-02:
Interesting because my run a command is firing off twice per motion event. So that's fun.

arturrocha, 2020-06-18:
Check file permissions, it needs to be executable.

JamBalaya56562, 2026-11-08:
Closing as answered above: putting the script where the user motionEye runs as can read and execute it, e.g. `/etc/motioneye/`, and giving its full path made it run. The command does not run in your desktop session, so a script that opens a browser has no display to open it on.

closed 2026-11-08 by JamBalaya56562