motioneye-project/motioneye #1348: Install problems with Raspian Buster

Pi-DD, 2019-07-01:
Trying to install Motioneye on fresh copy of Buster (ie not an upgrade from Stretch) I encounter the following problems....
`pi@DDhomepi6:~ $ sudo apt-get install libmariadbclient18 libpq5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libmariadbclient18 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libmariadb3

E: Package 'libmariadbclient18' has no installation candidate
`
Change to use alleged replacement....

> pi@DDhomepi6:~ $ sudo apt-get install libmariadb3 libpq5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libpq5 is already the newest version (11.4-1).
libpq5 set to manually installed.
The following additional packages will be installed:
  mariadb-common mysql-common
The following NEW packages will be installed:
  libmariadb3 mariadb-common mysql-common
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.

Works fine until dpkg of Motion...

`pi@DDhomepi6:~ $ sudo dpkg -i pi_stretch_motion_4.1.1-1_armhf.deb
Selecting previously unselected package motion.
(Reading database ... 135284 files and directories currently installed.)
Preparing to unpack pi_stretch_motion_4.1.1-1_armhf.deb ...
Unpacking motion (4.1.1-1) ...
dpkg: dependency problems prevent configuration of motion:
 motion depends on libavcodec57 (>= 7:3.2.9) | libavcodec-extra57 (>= 7:3.2.9); however:
  Package libavcodec57 is not installed.
  Package libavcodec-extra57 is not installed.
 motion depends on libavdevice57 (>= 7:3.2.9); however:
  Package libavdevice57 is not installed.
 motion depends on libavformat57 (>= 7:3.2.9); however:
  Package libavformat57 is not installed.
 motion depends on libavutil55 (>= 7:3.2.9); however:
  Package libavutil55 is not installed.
 motion depends on libmariadbclient18 (>= 5.5.36); however:
  Package libmariadbclient18 is not installed.
 motion depends on libswscale4 (>= 7:3.2.9); however:
  Package libswscale4 is not installed.

dpkg: error processing package motion (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (241-5+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 motion
`

Should I assume that for now install not possible on Buster, or is there a workaround somebody of my limited capabilities can manage? 

Thanks!

Pi-DD, 2019-07-01:
I’ve checked with the software list for Buster and it includes Motion 4.1.1 so I wonder whether simply installing that instead of the step in the installation instructions would suffice?

zagrim, 2019-07-02:
Obviously the motion package targetting Stretch doesn't like the changed dependencies in Buster (it's many times a pain to get a Debian(-ish) package to install on a "wrong" release). Please try with the motion package that's coming with Buster, at least that should run, and unless the Raspbian version has somehow incompatible build configuration, it might probably be just fine.

I don't myself have any yet experience on Buster (either Debian or Raspbian) so I might be missing something here, of course.

Pi-DD, 2019-07-02:
OK - I installed Motion 4.1.1-1 from the Buster package set and that seemed to work OK. Some dependencies were missing, but the install suggested using "apt -- fix-broken install" and that worked just fine.

Then moved to install Motioneye via pip.  jinja2 and pillow were already there, but the installation failed thus...

> pi@DDhomepi6:~ $ sudo pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Downloading https://files.pythonhosted.org/packages/00/ca/4dfc8d05f4783a8bf11b1ef1994c3adb4e49b6f478781464f648d5673389/motioneye-0.40.tar.gz (756kB)
    100% |████████████████████████████████| 757kB 87kB/s 
Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye) (2.10)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye) (5.4.1)
Collecting pycurl (from motioneye)
  Downloading https://files.pythonhosted.org/packages/ac/b3/0f3979633b7890bab6098d84c84467030b807a1e2b31f5d30103af5a71ca/pycurl-7.43.0.3.tar.gz (215kB)
    100% |████████████████████████████████| 215kB 142kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-NYFRZp/pycurl/setup.py", line 944, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-NYFRZp/pycurl/setup.py", line 606, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-NYFRZp/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-NYFRZp/pycurl/setup.py", line 233, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-NYFRZp/pycurl/

Interpreting that beyond my competence!  Any help appreciated!

kleini, 2019-07-02:
@Pi-DD please read error messages carefully. The cause of the issue is clearly mentioned there.

> Could not run curl-config: [Errno 2] No such file or directory

Some tool - curl-config - is not available in your system. Try to get it installed. When looking on its name it seems to be related to the curl client for server tool.

closed 2019-07-02 by kleini

zagrim, 2019-07-02:
In my Raspbian Stretch setup `/usr/bin/curl-config` originates in libcurl4-openssl-dev package, and I'd assume that also in Buster some libcurl*-dev package contains it. Too bad that Raspbian doesn't seem to have similar web interface for searching the repos as Debian has.

Pi-DD, 2019-07-02:
Thanks Marcus - I make no claims beyond "beginner" status, so I am groping my way to making this work!

I checked that curl was installed, it was. Did a bit of further research and concluded that maybe this step completing improperly was the problem (see above)...

`apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev`

despite running the fix-broken install. Found something on another forum that pointed to curl-config message being fixed by installing some of those elements, so ran 

`sudo apt install libcurl4-openssl-dev libssl-dev`

This worked just fine, completed OK. 

ran ...
`sudo pip install motioneye`
which then worked. 

Followed remaining installation steps as per Wiki and I now have a working Motioneye application. Not yet sure it is correctly identifying the camera as it does not seem to be giving me the options I get on other setups, but this is a start.

So, in summary -

- Motioneye will install on Raspbian Buster 
- in lieu of step 3 just install Motion 4.1.1-1 directly from the packages available for Buster
- if step 4 works with no errors, you may be OK
- if it does not you may need to install libcurl4-openssl-dev and libssl-dev again as above.
- then proceed with steps 5 onward. 

No guarantees! This is the result of trial and error!

GISGusting, 2019-08-12:
Thanks for al. Finally I managed to install MotionEye.
My steps with Raspbian Booser (as of Lite 2019-07-10) on Pi Zero W:

1. The same: `apt-get install ffmpeg v4l-utils`
2. libmariadbclient18 is replaced: `apt-get install libmariadb3 libpq5` (no idea what is it for, I just include)
3. Install motion from repo (4.1.1): `apt-get install motion`
4. The same works all right: `apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev`
5: To avoid fail, install pillow first: `apt-get install python-pillow` then motioneye: `pip install motioneye`

The rest of the steps are OK.

Hope this helps.

zagrim, 2019-08-12:
@GISGusting and @Pi-DD, thanks for providing and confirming the steps, hope someone with proper rights will update the wiki page (@ccrisan ?).

ccrisan, 2019-08-12:
Guys, I'll update the wiki asap.

Zciurus-Alt-Del, 2019-08-19:
I followed all of the steps GISGusting mentioned but when I do `pip install motioneye` it still fails with this error message:

root@raspberrypi:/home/pi# pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Could not find a version that satisfies the requirement motioneye (from versions: )
No matching distribution found for motioneye

GISGusting, 2019-08-23:
@Zciurus-Alt-Del Sorry, could you try `apt-get install motioneye` or `apt install motioneye` instead of pip? I'm afraid it is a package, not a Python module.

Zciurus, 2019-08-23:
On the [wiki page](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) for installation on raspbian it says to use `pip`. Nevertheless I did try `apt-get` and `apt` but both didn't work.

CRImier, 2019-08-23:
@Zciurus any chance the date in your OS is not set correctly? Looks like you have a `pip` problem.

Zciurus, 2019-08-23:
I actually moved away from using motioneye and instead just used ffmpeg because it did the job for the specific project and I'm no longer working on it and can't test it out at the moment.
However, yes, It is very well possible the date wasn't set correctly or something else made `pip` not work. My problem was **most likely not a motioneye issue**.

GISGusting, 2019-08-27:
> On the [wiki page](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) for installation on raspbian it says to use `pip`. Nevertheless I did try `apt-get` and `apt` but both didn't work.

Sorry @Zciurus I mixed up with `motion` :S

akingscote, 2020-02-29:
used pip3 instead of pip in GISGusting and managed to get going