motioneye-project/motioneye #1350: motioneye setting date failed !!

geoffp13, 2019-07-01:
Unable to set time zone either in expert settings or at command prompt. Also problems with motion detection being unreliable. Reloaded software 3 times - worked at intial setup with GMT but after updateing reverted to 1970 again. Motion detection worked sort of before changes to settings and then failed !! Looked through all previous user problems and found poor assistance non of which worked.
Using Raspberry Zero W with Motioneye motioneyeos-raspberrypi-20190427.img.
Checked log files:
Motion log file:
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)

Now you can see from the following time stamps that motioneye worked correctly at initial boot up but after changes made and reboot it reset date and lost motion.conf ??
2019-07-01 10:57:38: [motioneye]     INFO: bye!
1970-01-01 01:02:19: [motioneye]     INFO: hello! this is motionEye server 0.40
1970-01-01 01:02:23: [motioneye]     INFO: cleanup started
1970-01-01 01:02:23: [motioneye]     INFO: wsswitch started
1970-01-01 01:02:23: [motioneye]     INFO: tasks started
1970-01-01 01:02:24: [motioneye]     INFO: mjpg client garbage collector started

So much promise but then !!!

geoffp13, 2019-07-01:
Also tried zdump but does not exist !!
I also have a Pi 4 (latest version, just arrived) which I am delighted with and - so far - no issues at all. As a Linux user for many many years its a delight to be back again after a short sabatacle. ;-)
Incidently used a windows freeby software with the PZero camera and works a treat - no problems at all except I have to leave the windows machine on all night !! Not good.

geoffp13, 2019-07-02:
Well - still trying to get this F%$£&^G software to work. Finally get a set of pictures **immediatley after** changing settings then nothing when a double decker bus goes past !! This is so frustrating !! Still cannot get time to work correctly either.
This is the only setting that works briefly !!

[image]

Have redone the sd card and it works as long as you don't make any changes to settings. 
Can anyone help PLEASE  ;-)

zagrim, 2019-07-03:
Please go to [motionEyeOS project](https://github.com/ccrisan/motioneyeos) where you might get more appropriate help with this. This issue shouldn't be anything about the application motionEye but the overall system configuration.

JamBalaya56562, 2026-11-08:
Closing, as zagrim pointed you to motionEyeOS above: this was the motionEyeOS image of 20190427 on a Pi Zero W, where the time and the system settings are part of the OS. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos). If motion detection is still unreliable with a current motionEye, please open a new issue with `motion.log`.

closed 2026-11-08 by JamBalaya56562