motioneye-project/motioneye #1351: Camera not properly recognised on Raspbian Buster

Pi-DD, 2019-07-02:
See this item for background information....
https://github.com/ccrisan/motioneye/issues/1348#issue-462824686

I recognise this is “bleeding edge” since Buster very new and its base of Debian 10 not out for a few days!!,

Installation on Buster seems possible, as described in the link, but when installed

1. Version information is missing from General Settings
2. Camera is not fully recognised (locally attached Pi Noir) when it is recognised fully on Stretch. Running the camera direct from Raspivid works OK. 

See attached
[image]
[image]

First screenshot from Buster, second from Stretch. 
Note missing version info in general settings

In both cases camera configured as mmal camera followed by the only option offered of Videocore Camera. 

Buster version white balance never adjusts and no video resolution or other options appear.

ccrisan, 2019-07-02:
Enable advanced settings on Buster and you should be fine.

Pi-DD, 2019-07-03:
Calvin, thanks for that! User error not spotting it! 

Turning it on reveals detail but purple colour cast remains. Turning on Auto Brightness puts something into a loop of brightening and dimming that never settles to a specific state, all with a purple overcast. 

When I have time I’ll investigate further within the limits of my skill!

Pi-DD, 2019-07-03:
Turns out this is a known problem in the Pi AWB algorithm that was changed recently and does not handle the NOIR case. There is a workaround.

https://github.com/raspberrypi/firmware/issues/1167
https://www.raspberrypi.org/forums/view ... 3&t=241633

Sorry to have pointed this at the excellent MotionEye when (a) I made a user error in not noting Advanced Settings and (b) my adventure of Buster install gave me the mindset it was a MotionEye issue!

closed 2019-07-03 by Pi-DD