motioneye-project/motioneye #1352: Motioneye missing files thread0.conf in Pi Zero W

geoffp13, 2019-07-02:
After much frustration trying to get motioneye to work I finally found it needs a thread.conf for each camera which is a missing step in the build - probable affects all builds.
Add "thread-0.conf" for camera1 with the following then put it in /data/etc/ and all should be well ;-)
videodevice /dev/video0
stream_port 8081
Add thread-1.conf etc incrementing dev as necesary

geoffp13, 2019-07-02:
OK had a good play and all is fine although the tweeking can be time consuming. It recognises people, cars AND double decker buses now. Caveat to above if "thread-0.conf" makes no difference after reboot try "thread0.conf" ie: remove the hyphen.
ttfn

zagrim, 2019-07-03:
I suggest you provide the configuration files for motionEye and tell which versions of motion and motioneEye you're running (and for clarity, if it's actually motionEyeOS and not just a standalone installation of motionEye). Right now what you have stated sounds very confusing to me at least.
MotionEye should create thread-*.conf (in the older versions) or camera-*.conf (in the current release) automatically, with a hyphen in the filename.

zagrim, 2019-07-03:
Reading the other issue you've posted I see that you are indeed running motionEyeOS, which is not the same as standalone motionEye. At least I'm not at all familiar with how things are set up in MEOS, so it might be better if you took also this issue there.

JamBalaya56562, 2026-11-07:
Closing, as you reported that all is fine now. As zagrim noted above, this was on motionEyeOS, which is a separate project with its own setup.

closed 2026-11-07 by JamBalaya56562