motioneye-project/motioneye #1355: "Minimum Motion Frames" unit of measure: frames or seconds?

PieBru, 2019-07-03:
Hi,
IMO it would be better to measure "Minimum Motion" as seconds (or milliseconds) instead of frames.
This way the "Minimum motion" config parameter wouldn't be anymore logically linked to the "Video device / Frame rate" parameter, with easier maintenance over time.
Thanks,
Piero

jasaw, 2019-07-04:
It's more difficult than it sounds. The actual frame rate varies depending on your camera configuration. If your camera is set to automatic exposure mode, the actual frame rate will vary depending on lighting conditions. Poor lighting condition means longer exposure means lower frame rate.

To have minimum motion frames in milliseconds means to constantly adjust the number of frames used for motion detection. We may have to timestamp every frame in our image buffer so we always know how many frames to use for motion detection. I'm sure there are corner cases to think about in the implementation like what is the correct thing to do when frame rate drops to 1 or 2 fps, but minimum motion is set to something less than frame duration (like 100ms).

I like the idea. Can you please open this issue in [motion](https://github.com/Motion-Project/motion) project to see if motion developers have any appetite for this before we spend time implementing it? We want to push all motion modifications upstream because we don't have time to maintain non-standard set of features.

xenpac, 2019-10-10:
Interesting topic!
let say you have a config of 15 netcams.
to run them on 25 fps would somehow overload your LAN.
If you capture stills from each cam 3 times a second, well that might be just the lower end of motion detection.
maybe 5 stills per second would be a workable config.
so capturing stills instead of streams would be a solution to overcome LAN overload.

JamBalaya56562, 2026-11-15:
As @jasaw explained, motion counts `minimum_motion_frames` in frames, and with automatic exposure the real frame rate changes with the light, so motionEye can't turn seconds into a fixed number of frames for it. This would have to be added to motion itself (Motion-Project/motion), as he suggested. Closing as not planned here.

closed 2026-11-15 by JamBalaya56562