motioneye-project/motioneye #1356: Raspbian Wiki instructions

jorrinpollard, 2019-07-03:
Hello, I just installed this on the latest Raspbian Buster and ran into a few issues with the instructions [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). Namely `libmariadbclient18` not being a release candidate and `motion` having missing `libavcodec` dependencies after installation. I threw together a [provisioning script](https://gist.github.com/jorrinpollard/4fdf7cbe5f46c201ec719232e706fa7b) that you can put in there if you'd like. Awesome project by the way. Thanks.

ilovelinux, 2019-07-15:
Also `llibmariadbclient18` and `libpq5` are required only to enable MariaDB and PQ support. Do MotionEye uses these Motion features?

ppareit, 2019-07-25:
In Buster:

```
$ apt-version motion
motion:
  Geïnstalleerd: 4.1.1-1.1+b2
  Kandidaat:     4.1.1-1.1+b2
  Versietabel:
 *** 4.1.1-1.1+b2 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
        100 /var/lib/dpkg/status
```
So I think that it is no longer needed to wget the latest version of motion.

JamBalaya56562, 2026-10-26:
The per-distribution installation pages have been removed from the wiki (December 2025). There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. Closing, as the page this refers to no longer exists. If the README instructions do not work for you, please open a new issue.

closed 2026-10-26 by JamBalaya56562