motioneye-project/motioneye #1364: Execute Script after detecting Motion

RobinNowak, 2019-07-09:
Preliminary Docs
I confirm that I have read the CONTRIBUTING guide before opening this issue.
I confirm that I have read the FAQ before opening this issue.

0.40 motionEyeOS Version
4.0 Motion Version
9.9 Raspian (Stretched)
Raspberry Pi 3 (B+) Board Model
Raspberry Pi V2.1 Camera

Hello,
I want to run a command after a Motion is detected.
After reading a lot of differet issues i know that the file should be stored in /data/etc/example.sh
Unfortunatly i dont have this Directory.

The motioneye.conf file is saved in /etc/motioneye .

[image]

Ive installed motioneye os on my Raspberry pi (Raspberry 9.9 Stretched)

I tried putting different paths in to the Run a Command field but the script was never executed even though a Motion was detected and the script is doing its job after executed in the terminal.

[image]

(The file was made executable with chmod +x)

Anybody knows where I need to save the script or something else to solve my problem?

RobinNowak, 2019-07-09:
[image]

Is it even trying to run the command after detecting Motion?

Pi-DD, 2019-07-09:
Thanks for asking that! Struggling with the same problem. Wiki not clear. 
I had sort of deduced that because the "Monitoring Commands" seems to need to go in configuration folder (/etc/motioneye) these commands need to be in the same place, although these two paragraphs seem to contradict one another.....

*************
**Enabling A Monitoring Command**

motionEye _will look inside its configuration folder for executable files_ named monitor_[cameraid], where cameraid is the id of the camera on top of which the monitoring info will be displayed.

For example, on a setup using the default configuration, the _presence of the executable file /data/etc/monitor_1_ will determine motionEye to run the script every 1 second (by default), to read its standard output and to display it on top of the first camera frame.

**************

I have tried...

1. symlink in etc/motioneye/ pointing to script elsewhere
2. full directory path to the script in the web config tool
3. script in the etc/motioneye/ directory

none seem to work. File is executable by anyone.

Pi-DD, 2019-07-09:
I wanted to do the same thing, and agree the wiki is unclear, referring to "stored in /data/etc/example.sh".

However, I've managed to make it work OK, running a command both when motion is detected and when the video file creation is completed, by putting my bash script in /etc/motioneye/ just as you did. I am running all the same software levels as you - except you mention that you are running motioneyeos, I am running motioneye on Raspbian Stretch. Do you mean MotionEyeOS? 

My command script is executable by anyone, in case that is relevant. 

Looking at your log files there seem to be errors to do with config items not being recognised etc. Stupid question, but perhaps it is never getting to the point of detecting motion because it's not recognising the camera properly?

If you turn off the command, do you get still/video actually recorded when motion is detected? If you turn on "show frame changes" do you see that change when you move in front of the camera?

JamBalaya56562, 2026-11-07:
Closing as answered: as Pi-DD found above, a script in the configuration folder (`/etc/motioneye` on a standard install) runs on motion through the Run A Command option under Motion Notifications. Make it executable and enter its full path.

closed 2026-11-07 by JamBalaya56562