motioneye-project/motioneye #1365: Unable to access admin externally.

clrtech, 2019-07-11:
I have 2 cameras I can view side by side internally with the admin account. However after port forwarding I can only view one camera with the surveillance account.

trees-smoke, 2019-07-11:
Odd, my external setup works fine.
I have my home desktop reverse tunneled to a remote server via ssh, then that machine adds the TLS certs and proxies to the local port.
So, 127.0.0.1:8765 on the motionEye server reverse tunneled to 127.0.0.1:8765 on my remote server.  Then apache2 forwards any requests to /cams/ to 127.0.0.1:8765,
```
<Location "/cams/">
    ProxyPass http://localhost:8765/
    ProxyPassReverse http://localhost:8765/
</Location>

```
Does the admin account show both cameras? And you're certain that you port forwarded to the machine with both cameras and not a single camera feeding into MotionEye?

clrtech, 2019-07-11:
The admin account does show and everything is setup fine and works great internally. I can view the camera externally so the port forwarding is working... the big question is why can't I login with the admin account externally??

Pi-DD, 2019-07-11:
Precisely what message or messages do you get when you try to login with the admin account from external point?

clrtech, 2019-07-11:
It prompts for credentials with a pop up box saying the connection is not private and never lets a person in just cycles through repeatedly asking for user name & password until you enter the surveillance account info and then it shows the camera

[image]

No other cameras nothing to access the settings just a view of my room (blurred out its a little messy)
[image]

For the record the view of the room is awesome! The cameras work amazing.

Pi-DD, 2019-07-11:
That’s interesting - what browser is that from? Have you tried it with a different one? From a device with a different OS? The login panel looks nothing like the one I see on Safari (iOS) or Firefox (Win10) or Chromium on Raspbian. 

Maybe some browser setting is also involved? 

Your second screenshot also does not seem to show the top bar for MotionEye that I get even on the “surveillance” account

Before I set things up behind Nginx I could forward straight to my main MotionEye from outside just fine, but that was a while ago!  

If nothing else works maybe set it up behind Nginx running on the main MotionEye device - there are instructions in the Wiki and it’s not too difficult. It worked for me without setting up https as well and did not have the problem you are encountering.

clrtech, 2019-07-11:
Its the same everywhere this is firefox 
[image]

I remember the login screen looking the same externally as it is internally. I am not at home but it was the black Motioneye screen with the login information in the middle. This pop up was new. All of this behavior is the same across multiple devices and browsers. I even went as far as getting fresh SD cards and putting the latest OS on them and setting things up from scratch

Pi-DD, 2019-07-11:
This may be completely irrelevant, but.......

The two passwords are stored in etc/motioneye/motion.conf

The admin one is stored in some simple hashed form, the “surveillance” one in plain text. Take a look to see if yours is the same. 

I’m wondering if, for some reason far beyond my technical knowledge, something about your “external” route is failing to pass the admin password in a way that allows it to be compared with the stored form, whereas the “surveillance” one is?

Pi-DD, 2019-07-11:
This is what I see in all browsers and all OS....

Are we looking at the same thing? 🤨😏

[image]

Pi-DD, 2019-07-11:
I notice that in your screenshot it says “The site says”Motion””

This is another stupid question, but Are you sure you are not connecting to the “Motion” web interface rather than the “MotionEye” one? 

What port number are you using?

clrtech, 2019-07-11:
I am using the default internal ports that Motioneye sets and then I have tried 80, 8082 to test the port forwarding all allowed access but with this issue. The screenshot you put in is absolutely what I see internally and what I am not getting externally. Externally I get that pop up I posted.

Could you elaborate on the "Motion" web interface vs "MotionEye" this may be where my knowledge is lacking and the problem lies.

Pi-DD, 2019-07-11:
Motion is software that manages cameras and their various parameters. It is a command line tool generally, but does have a (simple!) web interface. MotionEye provides a better, more sophisticated web interface that can manage multiple cameras and includes control of Motion parameters. 

Re port forwarding.......

At a very simple level have you tried forwarding, in your router, some external port that is not a commonly used on (eg 7755) to port 8765 (The one MotionEye is accessed on) on the IP address of the “main” MotionEye machine ie the one that has its own camera and also points to the other MotionEye camera?

clrtech, 2019-07-11:
I am missing something. I changed the streaming on MotionEye 
[image]

Then changed my port forwarding to a random port to that IP and port and still have the same results
[image]

Pi-DD, 2019-07-12:
I'm sorry - I didn't explain myself clearly enough!

I assume that when you connect **internally** to your "main" MotionEye device you do so thus:-

http://my.local.ip.addr:8765 as per the "Installation" page in the wiki. That is the only port that will expose the control web pages to you, so you need to access that one externally as well.

- Go back to the web interface internally where you changed the port number for "Streaming Video". 
- Just to keep things tidy, change it back to whatever it was before.
- Turn "Video streaming" off for both the cameras, you don't need it.
- Leave the router forwarding as you set it in your post above.

When you access "externally" you should now see the web controls just as you do internally.

The "Video Streaming", as I understand it, is only for the situation in which you want your Motioneye device to send a video stream elsewhere, which is not what you are trying to do. This, again, as I understand it, is because for its own purposes Motioneye sends video to its web control pages in a particular way that other devices may not like. 
 
Hope that now works for you!

clrtech, 2019-07-12:
Thank you so much!!!! I didn't actually use the instructions in the Wiki my Linux skills are not great so I did a flash of the SD card with Etcher - so my internal stuff wasn't on port 8765. Once I found the port internally and put that in things fired off immediately. This was driving me nuts I greatly appreciate the help.

Pi-DD, 2019-07-13:
No problem! I’m very much a Linux beginner as well so I understand where you are coming from!

closed 2019-07-30 by clrtech