motioneye-project/motioneye #1366: "Preferences - "Layout Columns" not visible and does not work on Safari

Pi-DD, 2019-07-11:
I cannot change Layout Columns on Safari (iPad Air2, iPhone 6s, iOS 12.3.1) because it is not visible. Likewise FireFox on iOS. 

If I try on FireFox (Win 10) I can see it and change it, but the changes are not reflected in Safari.
First Picture from iOS Safari, second from Win10 Firefox

[image]

[image]

zagrim, 2019-07-12:
I think this is by design. A comment in the UI code says "always 1 column when in full screen or mobile" (here "mobile" means having display width less than or equal to 1200 pixels).

Pi-DD, 2019-07-12:
You are so right! The screenshots in the Wiki show it too, but are not super explicit.

My error of understanding!

Thanks

closed 2019-07-12 by Pi-DD