motioneye-project/motioneye #1370: Setting "white balance" value for RPi camera via MotionEye

Pi-DD, 2019-07-16:
Is there a way of setting this via any of the MotionEye settings? 

I can't find one and, as far as I can ascertain Motion does not contain this either, so I can't put it in "Additional Motion options".

Reason for asking - latest Pi firmware messes with NOIR cameras giving them a "purple haze". There is a temporary way around it to set camera to use old algorithm, BUT - forthcoming firmware to fix this will have a new "white balance" setting of  "greyworld" that you will need to set for NOIR camera to work. 

It would be useful to be able to readily set from MotionEye.

Pi-DD, 2019-09-03:
This can be done using Extra Motion Options thus......…

`mmalcam_control_params -awb greyworld`

(or other awb value) but "greyworld not yet supported in released version of Motion.

Closed, not Motioneye problem

closed 2019-09-03 by Pi-DD

mroberts77, 2020-11-09:
tried typing:
mmalcam_control_params -awb greyworld
per image:
[image]

didn't seem to make a difference, also tried 'tungsten' instead, no apparent effect: probably missed the way to add the info?