motioneye-project/motioneye #1371: motioneye fails to start after install

AndreVallestero, 2019-07-16:
Installing on RaspberryPi4, Raspbian Lite (debian buster) with Python 2.7 with Pip 2.7. 

systemctl status says `mtioneye.service: Failed at step EXEC spawning /usr/local/bin/meyectl: No such file or directory`
meyectl was installed to `~/.local/bin/meyectl` so I tried a symlink to `/usr/local/bin`meyectl` and the error I got was "Too many symbolic link layers".
Then I made a copy of meyectl to `/usr/local/bin` and I got the error `ImportError: No module named motioneye.meyectl`

AndreVallestero, 2019-07-16:
Solution:
Installing motioneye with pip still requires root. 
Either 
`sudo pip install motioneye`
or
`sudo su`
`pip install motioneye`

closed 2019-07-16 by AndreVallestero

iadegesso, 2020-06-03:
I did the install from the root user and meyectl has been installed in /usr/local/bin (pure Debian Buster)