motioneye-project/motioneye #1374: Extremely high CPU usage

MPK1, 2019-07-19:
Hi there,

I read a lot about CPU issues, but mine is absolutely not normal any more.
I run the latest motioneye docker container on my server (8 cores, 30GB RAM, SSD,...).
Motioneye is using the whole CPU (its on 100% all the time).
I'm using it with 26 RTSP cams, but without motion detection, just for live viewing.
There is nothing else running on the server, and if I stop motioneye the CPU usage drops down to 1%.

Does somebody know what I'm doing wrong?

Thank you very much!

zagrim, 2019-07-19:
Could you please tell in more detail which processes consume the CPU? Something like the output of ` top -bn 1 -o %CPU |head -20` would do.

MPK1, 2019-07-19:
top - 12:16:08 up 2 days,  3:04,  1 user,  load average: 51.79, 50.14, 47.83
Tasks: 174 total,   7 running, 167 sleeping,   0 stopped,   0 zombie
%Cpu(s): 10.2 us,  2.7 sy,  0.0 ni, 85.4 id,  0.0 wa,  0.0 hi,  0.2 si,  1.6 st
KiB Mem : 30886904 total, 26678856 free,  2581772 used,  1626276 buff/cache
KiB Swap:        0 total,        0 free,        0 used. 27849708 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
31181 root      20   0 5899464 0.984g  38168 S 720.0  3.3 139:53.67 motion
 8334 root      20   0 6014208 995144    336 R  40.0  3.2   0:01.55 ml20:Kameraserv
 8365 marcel    20   0   42832   3520   2952 R   5.0  0.0   0:00.02 top
    1 root      20   0   57064   6776   5212 S   0.0  0.0   0:14.04 systemd
    2 root      20   0       0      0      0 S   0.0  0.0   0:00.12 kthreadd
    3 root      20   0       0      0      0 S   0.0  0.0   7:08.42 ksoftirqd/0
    5 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 kworker/0:0H
    7 root      20   0       0      0      0 S   0.0  0.0   4:10.93 rcu_sched
    8 root      20   0       0      0      0 S   0.0  0.0   0:00.00 rcu_bh
    9 root      rt   0       0      0      0 S   0.0  0.0   0:00.71 migration/0
   10 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 lru-add-drain
   11 root      rt   0       0      0      0 S   0.0  0.0   0:02.00 watchdog/0
   12 root      20   0       0      0      0 S   0.0  0.0   0:00.00 cpuhp/0

zagrim, 2019-07-19:
I think you'd better take this to [motion project](https://github.com/Motion-Project/motion), since it's very likely related to how motion (the application that really does all the video processing for motionEye) handles the streams from your cameras.

JamBalaya56562, 2026-11-10:
As found above, the CPU goes to Motion (720% in the `top` output), which decodes all 26 RTSP streams even for live view only. Lower resolutions or the cameras' sub-streams, and a lower frame rate, reduce it; with only live viewing needed, pointing the viewer at the cameras' streams directly avoids the decoding. Closing as answered.

closed 2026-11-10 by JamBalaya56562