motioneye-project/motioneye #1376: Kernel Oops after a few days of operation

miknuc, 2019-07-20:
Hi,

after some days of operation I encountered a kernel Oops:
[Mon Jan  5 01:17:50 1970] pgd = d56d8000                                                              
[Mon Jan  5 01:17:50 1970] [0000014b] *pgd=16a11831, *pte=00000000, *ppte=00000000                     
[Mon Jan  5 01:17:50 1970] Internal error: Oops: 17 [#1] ARM                                           
[Mon Jan  5 01:17:50 1970] Modules linked in: ipv6 spidev brcmfmac snd_bcm2835(C) brcmutil snd_soc_bcm2835_i2s regmap_mmio evdev snd_soc_core joydev snd_compress snd_pcm_
[Mon Jan  5 01:17:50 1970] CPU: 0 PID: 6252 Comm: modprobe Tainted: G         C      4.14.79 #1                                                                           
[Mon Jan  5 01:17:50 1970] Hardware name: BCM2835                                                                                                                         
[Mon Jan  5 01:17:50 1970] task: d6b38de0 task.stack: d5a52000                                                                                                            
[Mon Jan  5 01:17:50 1970] PC is at ret_fast_syscall+0x0/0x28                                                                                                             
[Mon Jan  5 01:17:50 1970] LR is at vfs_read+0x150/0x168                                                                                                                  
[Mon Jan  5 01:17:50 1970] pc : [<c000fd80>]    lr : [<c015af00>]    psr: 60000013                                                                                        
[Mon Jan  5 01:17:50 1970] sp : d5a53fa8  ip : d5a53f50  fp : 00000000                                                                                                    
[Mon Jan  5 01:17:50 1970] r10: 00000000  r9 : d5a52000  r8 : c000ff24                                                                                                    
[Mon Jan  5 01:17:50 1970] r7 : 00000003  r6 : b6ff23ea  r5 : 01d9d850  r4 : 01d9d850                                                                                     
[Mon Jan  5 01:17:50 1970] r3 : 00000000  r2 : 00056000  r1 : 00000000  r0 : 00000400                                                                                     
[Mon Jan  5 01:17:50 1970] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user                                                                              
[Mon Jan  5 01:17:50 1970] Control: 00c5387d  Table: 156d8008  DAC: 00000055                   
[Mon Jan  5 01:17:50 1970] Process modprobe (pid: 6252, stack limit = 0xd5a52188)                                                                                    
[Mon Jan  5 01:17:50 1970] Stack: (0xd5a53fa8 to 0xd5a54000)                                
[Mon Jan  5 01:17:50 1970] 3fa0:                   01d9d850 01d9d850 00000000 b6ff2000 00000400 00000000
[Mon Jan  5 01:17:50 1970] 3fc0: 01d9d850 01d9d850 b6ff23ea 00000003 01d97728 00000000 00000016 00000000
[Mon Jan  5 01:17:50 1970] 3fe0: 00000000 be9e9984 b6ebd590 b6f1295c 60000010 00000000 00000000 00000000
[Mon Jan  5 01:17:50 1970] Code: d3a00001 e89da800 00000000 00000000 (e5ad0008)                         
[Mon Jan  5 01:17:50 1970] ---[ end trace e71fe3bb3a04c8e1 ]---                                      

[root@meye-f6d19c14 log]# uname -a
Linux meye-f6d19c14 4.14.79 #1 Sat Jan 19 14:17:41 UTC 2019 armv6l GNU/Linux

System is a Zero W:

[root@meye-f6d19c14 log]#  cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000f6d19c14
[root@meye-f6d19c14 log]# cat /proc/device-tree/model
Raspberry Pi Zero W Rev 1.1

I encountered this before on a much higher frequency and did a fresh reinstall. Didn't help too much....
I am not experienced with kernel panics,. Is this a hardware related issue or a bug in Motioneye?

Any suggestions?
Best Regards
Michael

jasaw, 2019-07-22:
Try lowering the frequency to 950MHz: https://github.com/raspberrypi/linux/issues/2555#issuecomment-394105250

miknuc, 2019-08-05:
After some days of testing: this seems to solve the issue. 
Thank you for the quick support!

closed 2019-08-05 by miknuc